Moddable-OpenSource/moddable #333: JNI like interface to link existing native libraries

MarsPlanet, 2020-03-04:
This is not an issue but want to know if there is any documentation which describes how to link existing native libraries

dckc, 2020-03-12:
FFI is pretty extensively documented: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md

is there something else you had in mind?

MarsPlanet, 2020-03-18:
Looking for a simple "Hello World" example on how to integrate existing library. Thank you for link, will read through and see if it helps.

dckc, 2020-03-18:
I think [examples/js/xs-in-c](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/js/xs-in-c) is intended to be the "Hello World" of FFI.

phoddie, 2020-05-13:
@dckc correctly references the XS in C documentation. This interface is used for all the native modules in the Moddable SDK, so there are tons of examples available to use as a starting point.

closed 2020-05-13 by phoddie