Moddable-OpenSource/moddable #335: XS incorrectly interprets zero when used as a multiplication factor

jugglinmike, 2020-03-05:
When one or more factors is zero, the multiplication operator should return a zero value whose sign reflects the sign of all factors. XS version 10.0.0 erroneously ignores the sign in this case:

    $ ~/.jsvu/xs -e 'print(1 / (0 * -1) < 0)'
    false
    $ ~/.jsvu/xs -e 'print(1 / (-1 * 0) < 0)'
    false

This behavior was observed using [the `xst` binary](https://github.com/Moddable-OpenSource/moddable-xst) as provided by [jsvu](https://github.com/GoogleChromeLabs/jsvu).

    $ ~/.jsvu/xs -v
    XS 10.0.0
    $ uname -a
    Linux bruce 4.15.0-88-generic #88-Ubuntu SMP Tue Feb 11 20:11:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

phoddie, 2020-08-26:
This only occurs when both operands are integers.  These work as expected:

```
$ xst -e 'print(1 / (0.1 * -1) < 0)'
true
$ xst -e 'print(1 / (0 * -1.1) < 0)'
true
```

The problem is that the integer multiply optimization does not account for this special case.

phoddie, 2022-06-06:
Fixed when `mxMinusZero` is defined for the build (and `xst` will set `mxMinusZero` moving forward to prioritize conformance over performance).

jugglinmike, 2022-06-10:
Sounds good!

Dunno if the "please verify" label is addressed to me, but I don't have the bandwidth to build XS, so I won't be able to verify behavior controlled by build flags.

phoddie, 2022-06-11:
Hey, @jugglinmike. We usually leave it to the person reporting the issue to verify it. That's intended as a courtesy, not an obligation We can close this one out.

Just FYI – since `xst` is about testing conformance, all future `xst` binaries we poste will have `mxMinusZero` set so this issue will not occur there.

closed 2022-06-11 by phoddie

jugglinmike, 2022-06-16:
Sounds good. Congrats on fixing this one!