Moddable-OpenSource/moddable #338: Promise.all("string") and Promise.any("string") should not reject

rwaldron, 2020-03-12:
Both `Promise.all("string")` and `Promise.any("string")` are valid and should not result in a rejection: 

```
$ xs -e "Promise.any('').then(() => {}, print)"
TypeError: Promise.any: iterable is no object

$ xs -e "Promise.all('').then(() => {}, print)"
TypeError: Promise.all: iterable is no object
```

jugglinmike, 2020-03-13:
Same goes for `Promise.allSettled`:

    $ ~/.jsvu/xs -e 'Promise.allSettled("").then(null, print)'
    TypeError: Promise.allSettled: iterable is no object

On the other hand, XS correctly interprets primitive string values as valid iterables in all other contexts: `Array.from`, spread in arguments lists, spread in array initializers, for..of, delegating `yield`, array destructuring, and the `Set` constructor. This appears to be limited to the Promise static methods.

phoddie, 2020-05-13:
This is no longer reproducible. It likely got fixed in the XS 10 work. To test, I tried:

```javascript
async function test() {
	trace("a\n");
	await Promise.all("string");
	trace("c\n");
	await Promise.any("string");
	trace("d\n");
	await Promise.allSettled("")
	trace("e\n");

}

test();
trace("b\n");

```
That outputs:

```
a
b
c
d
e
```

closed 2020-05-13 by phoddie