Moddable-OpenSource/moddable #340: JSON.stringify throws cyclic value! (in preload module)

takeru, 2020-03-20:
JSON.stringify throws cyclic value! in preloaded module.

If remove "hello" from "preload", it is work well. 

I can see "cyclic value!" with esp32 and mac.

==================

manifest.json
```manifest.json
{
  "include": [
    "$(MODDABLE)/examples/manifest_base.json"
  ],
  "modules": {
    "hello": "./hello",
    "*": [
      "./main"
    ]
  },
  "preload": [
    "hello"
  ],
  "resources": {}
}
```

main.js
```main.js
import Hello from "hello";
const h = new Hello();
h.hey();
```

hello.js
```hello.js
export default class Hello {
  hey() {
    trace(JSON.stringify(data));
  }
};

const data = [
  {
    "a": 0,
    "b": 1
  },
];
```

phoddie, 2020-05-13:
Yes, the implementation of JSON serialization relies on being able to set a temporary bit in the each object to detect cycles. This is not possible for objects stored in flash. The workaround, as you note, is to ensure the object is in RAM.

It is theoretically possible to implement a different algorithm for JSON serialization that works for objects stored in flash. The most straightforward solution for that would require more memory, more code, and run more slowly, which isn't appealing. We are open to suggestions.

dckc, 2020-10-13:
How about checking whether an object is in flash before setting the visited bit?

This assumes
  - there's a flag for flash / ROM objects
  - cycles in ROM / flash are impossible (or: are prohibited by the API)

phoddie, 2020-10-13:
It isn't quite that easy.

- The `XS_LEVEL_FLAG` is used both to detect cycles and to indicate a slot is in flash. It might be possible to eliminate that overlap, but flag space is limited and using more memory isn't a good option.
- Cycles that involve flash are possible because the aliasing mechanism allows adding and modifying properties of objects in flash.

phoddie, 2022-06-02:
We've updated `JSON.stringify` to throw a "read only value" error instead of the incorrect and misleading "cyclic value".

With that, I'm going to close this issue. It would be nice to be able to use `JSON.stringify` with preloaded objects. Doing that seems like more work (and code) than is justified by the functionality. If there is a strong need identified for supporting `JSON.stringify` of preloaded objects, we can revisit this (preferably in a new issue).

closed 2022-06-02 by phoddie