Moddable-OpenSource/moddable #342: To-be-set PWM value checked against freq_hz in esp32/pwm.c

jernst, 2020-03-23:
I'm looking at [this line](https://github.com/Moddable-OpenSource/moddable/blob/f2e2ed271fca035f756b2b41c6e736e1a4cc8f6f/modules/pins/pwm/esp32/pwm.c#L128):
```
if ((value < 0) || (value >= gTimer.freq_hz))
		xsRangeError("bad value");
```
Why would the to-be-set value be compared against the PWM timer frequency? There is a relationship between resolution and frequency [according to IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html#ledc-api-supported-range-frequency-duty-resolution) but it appears more complex and it should affect the less-significant bits in the value, not the range I think.

The frequency is currently hard-coded to 1024Hz, as it seems, which happens to be the same as the end of the permitted range for the value, but that's a coincidence as far as I can tell.

andycarle, 2021-02-09:
@jernst Thank you for the report, and apologies for letting it sit for so long. As it happens, I fixed this in f4413a7 while dealing with another issue. You are correct: the old implementation worked by coincidence. Fixed!

andycarle, 2021-02-13:
I'm going to go ahead and close this out. Please let me know if there are any additional issues around PWM!

closed 2021-02-13 by andycarle