Moddable-OpenSource/moddable #343: Websocket: "Partial header!!"

mrpeu, 2020-03-27:
I am trying to have a webserver offering a websocket connection; to get a 2-way connection with a web browser. I've successfully used the HTTP server. A Websocket also instantiates successfully. Then I try to connect (from a web browser's devtools): ```let ws = new WebSocket("wss://192.168.178.52:81", "json");
ws.onopen = e => console.log("ws.onopen", e);```
And the server "traces": ```Partial header!!```
from https://github.com/Moddable-OpenSource/moddable/blob/4a6c84a7fa4ca3b0593243d90e7a1edf9c22f362/modules/network/websocket/websocket.js#L319
The network stacks go a little over my head so I might be doing something wrong. I figured I'd ask before starting to read deeper ^^

phoddie, 2020-05-12:
That's interesting. It isn't an error. It is an untested code path. The trace is there to see when it actually happens.

Did the connection work for you? If so, I should remove the trace since the path is now tested. ;)

mrpeu, 2020-05-16:
I left out a crucial information. When initiating the connection, the "partial header" message pops on the server, and then the browser/client times out with "WebSocket opening handshake timed out".
I am going to look if this handshake is supposed to be done by your lib or if I have to manually do something here.

phoddie, 2020-05-22:
Not sure what is going on. We have successfully used the WebSocket server to communicate with web browsers.

mrpeu, 2020-05-23:
It works! I was trying to connect to `wss://...` But `ws://...` works. I do no need a secure connection for my current applications so that is fine. I will have to look into it later, at least out of curiosity.

Thanks again for the response, I'm learning a lot.

closed 2020-05-23 by mrpeu