Moddable-OpenSource/moddable #346: Sonoff S20 (outlet)

mrpeu, 2020-03-30:
I was hoping to flash an outlet Sonoff S20 (ESP8266) as simply as the moddable_three but i cannot get it to even blink. As it is a simple outlet I assume that there is no need for specific drivers like in the Sonoff B1 example, as they are about the driving the LEDs. But uploading some blinker code is unsuccessfull, same for some HTTP server. These codes (with updated pins refs) work well on the moddable_three.

The upload process runs normally, with no warning nor error. The debugger does not attach but i can imagine that's to be expected, I have no idea.

**Is there something else to be aware of?**

Any hint how I could figure out what's going wrong?

PS: I am pretty sure of the pins I am using because flashing Tasmota works well.

phoddie, 2020-05-12:
Is the Sonoff S20 using a normal ESP8266 or the slightly different 8285? If the latter, you need to specify that in your build target to get the appropriate linker map:

```
mcconfig -d -m -p esp/8285
```

FWIW - details about using the Moddable SDK on the Sonoff B1 lightbulb are on our [blog](https://blog.moddable.com/blog/hacking-sonoff-b1/). Perhaps something there will be useful for the S20. Please do let us know what you find. It would be great to get the Moddable SDK running on the S20.

mrpeu, 2020-05-19:
At least this Sonoff S20 has an ESP8266.
I've already seen your blog post yes, and demo and youtube: great showcase of the simplicity of the process and great teaser to the potential of the concept :)

wilberforce, 2020-05-20:
> @wilberforce I see that you have experience with Espruino and Sonoff. Have you tried XS on Sonoff S20?
No - I don't have an S20. I have got a webserver running on a sonoff original using the Moddable esp/8285 driver.

You didn't mention what OS are using. If you connect to the Sonoff with a terminal programme - rather than xsbug, you might see some helpful messages. On windows you can use putty.

mrpeu, 2020-05-20:
Windows as well.

Thank you! I have to discover how that works; that is an exciting idea.

mrpeu, 2020-05-20:
The Putty window stays blank, even when I press the onboard button.

(Gibberish is printed when I try the same with the WemosD1. So I'll look for the correct setup to get useful serial output. But I guess it won't help with the mute S20)

mrpeu, 2020-05-20:
Is there a way to see some boot sequence. Putty only connects once the ESP is connected to my computer, so I cannot get any message that would be sent early. Is there a way for Putty or another serial terminal to be ready whenever something happens on the COM port?

phoddie, 2020-05-22:
The ESP8285 seems to require a little extra time to establish the serial connection.

I'm not sure if this will help but... in the Moddable SDK we add a 1000 ms delay on the esp/8285 target using [`modPrelaunch`](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/platforms/esp/xsHost.c#L711). That is enough time to allow the connection to settle so that the xsbug connection can be reliably established. The delay is configured in the [esp/8285 manifest](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp/targets/8285/manifest.json#L14).

mrpeu, 2020-05-23:
Hmm. That did not change anything either. I'm running out of ideas of random things to try. Last thing will be to order another one, in case this one is damaged. I'll update this issue when done and will close it for now.

Thank you for the responses 👍

mrpeu, 2020-07-02:
'Got it! The serial interface (CP2102) I am using has its VCC to 5V. When I use 3V it works.
With 5V, the upload looks all good, but the program never starts I think; at least not the webserver I'm uploading. When using 3V it works well.
One catch compared to the usual workflow I am getting used to on the WemosD1mini is that the "Hard resetting via RTS pin..." does not work on the S20. It has to be unplugged and replugged.
- So it is plug with pressing the button to enter boot mode
- Upload
- unplug and replug (or plug to power outlet, whatever gives it juice to boot normally)

PS: I have no electronic education, so this may all be tainted with false assumptions. Take it as naive observations.

phoddie, 2020-07-08:
Glad you were able to figure out the problem!

mrpeu, 2020-07-25:
My guess was wrong. I've ordered new s20 and they are behaving like the first one I tried. They are of different revisions. The v1.3 accepts Tasmota but not XS, and that's the  issue. The v2.1.0 simply worked.
Here's some more decriptions of these 2 boards:

S20_EU_V1.3 2017-2-16 (as written on the board) with an ESP8266EX 062018 TUB0PH6Y81
[image]

S20 UK/EU V2.1.0 2018-06-19 (as written on the board) with an ESP8266EX 182018 TUB0PHSH89
[image]

Next, I'm going to search for the differences that might impact me.

phoddie, 2020-09-11:
If I understand the thread above, you are programming the S20 with an ESP8266 target, not 8285. I recently committed changes to use slower SPI settings by default for ESP8266 because some boards don't work with the faster settings.

https://github.com/Moddable-OpenSource/moddable/blob/a000d3c3b154f9dfe6dd17d5fbdfea770e5e1428/tools/mcconfig/make.esp.mk#L57-L58

When it fails, the symptom is that they flash program correctly (always done at slow speed) but they do not run the software after restarting. That's what you appear to report.

So... if you update and try again, perhaps it will work.

> FWIW - The Moddable One and Moddable Three device targets (both ESP8266) default to full speed SPI because always shipped them with full speed capable modules. The performance difference is quite significant.

https://github.com/Moddable-OpenSource/moddable/blob/a000d3c3b154f9dfe6dd17d5fbdfea770e5e1428/build/devices/esp/targets/moddable_one/manifest.json#L6-L9

mrpeu, 2020-09-26:
Yes, you understood the issue perfectly. Unfortunately the slower speed is not solving it. The upload still looks good, but on reboot of the device, nothing happens.
I have tried the same simple application (blinking GPIO 12 and 13) on a similar device (the Sonoff Basic) successfully. And Tasmotizer can still be uploaded to this S20.
That makes me hopeful that there must be a configuration in the upload process that will work for XS 🤞

phoddie, 2020-09-26:
OK, so what are the flash settings the "Tasmotizer" build uses -- both the mode and speed?

mrpeu, 2020-09-27:
While looking for this information,

I just tried to upload the binary with esptool.py. Same: tasmota.bin boots, but not my main.bin from XS.

phoddie, 2020-09-28:
I wonder if there is some difference in the firmware (second stage  boot loader). The Moddable SDK uses [boot_v1.7.bin](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/components/esp8266/firmware) from the ESP8266 RTOS project. Maybe that is incompatible with the chip in the S20 somehow. Do you know what second stage boot loader is used by Tasmotizer?

mrpeu, 2020-09-28:
I'm only finding references to v1.4 and 1.5 in an older repository.

I think I have now tried with boot_v1.6 by modifying mcconfig/nmake.esp.mk and recompiling (makefiles\win\build) but that hasn't changed the s20 behavior.

I am continuing to try to understand how Tasmota is built.

closed 2022-01-27 by mrpeu