Moddable-OpenSource/moddable #348: ESP8266 and servo

mrpeu, 2020-04-14:
Wemos D1 mini is correctly getting some XS love. I can instantiate a basic web server, blink a LED on a GPIO. I'd like to move a servo now but the compilation is failing to access some libraries:

`### Error: '$(ARDUINO_ESP8266)/libraries/Servo/src/esp8266/*': directory not found!`

There are source files where the "getting started" doc indicates them to be (`C:\Users\mrpeu\esp\esp8266-2.3.0\libraries\Servo\src\esp8266`).

This error is the only output line from running `mcconfig` for the servo-sweep example.

mrpeu, 2020-04-19:
I guess this is just a path issue but I can not figure it out yet.

phoddie, 2020-05-12:
Apologies for the delayed reply. I just built the servo-sweep example for ESP8266 without an error:

```
cd $MODDABLE/examples/pins/servo-sweep
mcconfig -d -m -p esp
```
From your paths above, I am guessing you may be building on Windows. Perhaps `$(USER)` isn't resolving in the manifest there?

In `$MODDABLE/modules/pins/servo/manifest.json` try replacing `$(USER)` with its value. So, go from this...

	"build": {
		"ARDUINO_ESP8266": "/Users/$(USER)/esp/esp8266-2.3.0/",
	},

...to this...

	"build": {
		"ARDUINO_ESP8266": "/Users/mrpeu/esp/esp8266-2.3.0",
	},

(It looks like there is an unnecessary trailing slash on `ARDUINO_ESP8266`. I removed that in the above as well.)

Good luck.

mrpeu, 2020-05-18:
Yes! Hardcoding the value `$(USER)` solve this issue 👍 
Thank you for responding at all! Very appreciated.

closed 2020-05-18 by mrpeu

mrpeu, 2020-05-18:
PS: Since it is the only instance of `$(USER)` I see in the repo, I won't dig more what goes wrong with it.