Moddable-OpenSource/moddable #349: "/bin/sh: 1: set: Illegal option -o pipefail" Compile error on Ubuntu/ESP32

meganetaaan, 2020-04-16:
* Ubuntu 18.04
* Moddable 209f93e
* ESP-IDF v3.3.2

When I tried to build with the latest setup, I got an error below.

```sh
$ mcconfig -m -p esp32

...

Project build complete. To flash, run this command:
../../../../../../../esp32/esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --after hard_reset write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x1000 ../idf/bootloader/bootloader.bin 0x8000 ../idf/partition_table/partition-table.bin 0x10000 ../idf/xs_esp32.bin
or run 'idf.py -p (PORT) flash'
/bin/sh: 1: set: Illegal option -o pipefail
/home/sisikawa/Projects/moddable/build/tmp/esp32/release/neomatrix/makefile:472: recipe for target 'all' failed
make: *** [all] Error 2
```

This is caused by setting pipefail option in [tools/mcconfig/make.esp32/mk](https://github.com/Moddable-OpenSource/moddable/commit/11c20f7c861e43c59b1b57310e4a8ed819cb3cba). `sh` of Ubuntu is symlinked to `dash` which does not have pipefail option.

I think some workaround needed.

andycarle, 2020-04-17:
Thank you for the report—I'll take a look.

andycarle, 2020-04-17:
@meganetaaan I've pushed a fix for this issue to our local repository: it will be reflected in our next Open Source push. 

In the meantime, you can replace line 330 of [make.esp32.mk](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcconfig/make.esp32.mk) with:
`	cd $(PROJ_DIR) ; bash -c "set -o pipefail; $(DEPLOY_CMD) | tee $(PROJ_DIR)/flashOutput"`

Sorry for the confusion. Also, I have seen some Linux installations today have trouble with DEBUGGER_SPEED 460800. When that happens, the build and flash work correctly, but xsbug does not connect. If you see that, please try `export DEBUGGER_SPEED 921600` before your build.

andycarle, 2020-04-27:
This fix has been pushed. Please let me know if you have any other issues.

closed 2020-04-27 by andycarle