Moddable-OpenSource/moddable #350: After last update esp32 won't connect to xsbug after flash

diogoviannaaraujo, 2020-04-16:
After last update xsbug does not connects to running session on esp32.
When running the same example on Mac platform everything works fine.
Tested with helloworld example.

andycarle, 2020-04-17:
Hi @diogoviannaaraujo, apologies for the inconvenience. Let's get you back up and running as quickly as we can. 

I agree that the DEBUGGER_SPEED is the most likely culprit here. Another possibility is that we're incorrectly picking up your UPLOAD_PORT when starting serial2xsbug.

Could you do me a favor and try a build with verbose output? Just `export VERBOSE=1` before your build. If you could post a screenshot or log of the last bit of output from that (as you did above) that would help me a lot in sorting out the issue.

Thank you,
&nbsp;- Andy

phoddie, 2020-05-12:
Closing due to inactivity. If the problem persists, please re-open with additional details.

closed 2020-05-12 by phoddie

diogoviannaaraujo, 2020-05-12:
Hi, sorry for not returning on this, the problem was I was using `/etc/tty.SLABS` it all went nice when I used `/dev/cu.SLABS`.

What is weird is that I've been using TTY with moddable for some years, maybe is something with the ids upgrade.