Moddable-OpenSource/moddable #352: Build Issue on Mac

msaksena, 2020-04-18:
I am trying to build the Moddable SDK on my MAC and ran into the following build issue. 

For context, I am on a Mac. My shell is bash. I did a 'git pull' followed by running make (after a removing build/bin and build/tmp) in $(MODDABLE)/build/makefiles/mac where $(MODDABLE) is set to the root of the Moddable SDK. 

This runs into the following error: 

$make
......
# tools debug : cc xsmc.c
# tools debug : cc xsre.c
# tools debug : cc xsa.c
# tools debug : cc xsc.c
# tools debug : cc xslBase.c
# tools debug : xsc commodettoBitmap.js
make[1]: /Users/saksemc/moddable/build/bin/mac/debug/xsc: Input/output error
make[1]: *** [/Users/saksemc/moddable/build/tmp/mac/debug/tools/modules/commodetto/Bitmap.xsb] Error 1
make: *** [debug] Error 2
$

If I run make again, it builds some more and then fails. 

$make
make -f /Users/saksemc/Desktop/whirlpool/software/moddable/xs/makefiles/mac/xsc.mk
make -f /Users/saksemc/Desktop/whirlpool/software/moddable/xs/makefiles/mac/xsid.mk
make -f /Users/saksemc/Desktop/whirlpool/software/moddable/xs/makefiles/mac/xsl.mk
make -f serial2xsbug.mk
make -f simulator.mk
make -f tools.mk
# tools debug : xsc commodettoBitmap.js
# tools debug : xsc commodettoBMPOut.js
# tools debug : xsc commodettoBufferOut.js
# tools debug : xsc commodettoColorCellOut.js
# tools debug : xsc commodettoConvert.js
# tools debug : xsc commodettoParseBMF.js
# tools debug : xsc commodettoParseBMP.js
# tools debug : xsc commodettoPixelsOut.js
# tools debug : xsc commodettoPoco.js
# tools debug : xsc commodettoReadJPEG.js
# tools debug : xsc commodettoReadPNG.js
# tools debug : xsc commodettoRLE4Out.js
# tools debug : xsc file.js
# tools debug : xsc buildclut.js
# tools debug : xsc colorcellencode.js
# tools debug : xsc compressbmf.js
# tools debug : xsc image2cs.js
# tools debug : xsc mcconfig.js
# tools debug : xsc mclocal.js
# tools debug : xsc mcmanifest.js
# tools debug : xsc mcrez.js
# tools debug : xsc png2bmp.js
# tools debug : xsc resampler.js
# tools debug : xsc rle4encode.js
# tools debug : xsc tool.js
# tools debug : xsc wav2maud.js
# tools debug : xsc bles2gatt.js
# tools debug : xsid commodettoBitmap.c
make[1]: /Users/saksemc/moddable/build/bin/mac/debug/xsid: Input/output error
make[1]: *** [/Users/saksemc/moddable/build/tmp/mac/debug/tools/commodettoBitmap.c.xsi] Error 1
make: *** [debug] Error 2

If I keep running make repeatedly after each failure, it eventually builds everything.

andycarle, 2020-04-19:
Hi @msaksena, I'm sorry to hear that you're having trouble with building the Moddable SDK tools. Let's see if we can sort out what's going wrong here.

My understanding of that type of input/output error is that it is usually more about the state of the computer the build is running on and less about what exactly is being built. For instance, that problem can be caused by disk defects, permission errors, or antivirus software. 

It is probably worth investigating those potential causes one a time. I would start by disabling antivirus software momentarily and see if that helps any. Permissions issues could potentially be resolved by getting a fresh clone of the Moddable SDK. And as for general disk defects... you could try the macOS disk utility to check, but if you are not experiencing any other problems, that seems like a stretch.

If none of that helps, could you please attach a full log of the build output to this thread? Perhaps there's something more subtle that I'm missing from the snippet above.

Cheers,
&nbsp;- Andy

msaksena, 2020-04-21:
Andy: Thanks for your response. I think it was because I had an old copy of the Moddable SDK and the PATHS were a little messed up. Closing this.

closed 2020-04-21 by msaksena