Moddable-OpenSource/moddable #354: undefined reference to `xs_digital_monitor_get_pwm_freq' esp32 build

wilberforce, 2020-05-02:
After updating to the latest code, I have build issues for ESP32 target:

```
C:\Users\rhys\source\repos\www-m5-mash>mcconfig -d -m -p esp32/m5stick_c 
Checking Python dependencies...
Python requirements from C:\Users\rhys\esp32\esp-idf\requirements.txt are satisfied.
Running ninja in directory C:\Users\rhys\Projects\moddable\build\tmp\esp32\m5stick_c\debug\idf
Executing "ninja all"...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable xs_esp32.elf
FAILED: xs_esp32.elf
cmd.exe /C "cd . && C:\PROGRA~1\ESPRES~1\ESP-ID~1\tools\bin\XT1FE9~1.EXE  -mlongcalls  -nostdlib @CMakeFiles/xs_esp32.elf.rsp  -o xs_esp32.elf  && cd ."
xs_esp32.a(mc.xs.o):(.rodata.gxBuilders4+0x38): undefined reference to `xs_digital_monitor_get_pwm_freq'
xs_esp32.a(mc.xs.o):(.rodata.gxHeap+0xddd8): undefined reference to `xs_digital_monitor_get_pwm_freq'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
NMAKE : fatal error U1077: 'python' : return code '0x2'
```

thanks.

mkellner, 2020-05-02:
My apologies. I checked a fix in an hour ago. Please update.

wilberforce, 2020-05-02:
https://github.com/Moddable-OpenSource/moddable/commit/928e4e6293ab3b8111442a653f0f36419c8c9e6c#diff-deed9e7c75926eda4ce5067ac67cf32a

Thanks - builds now

closed 2020-05-02 by wilberforce