Moddable-OpenSource/moddable #357: moddable_two won't retain bootloader mode

mikehedman, 2020-05-08:
When I run `mcconfig -d -m -p esp32/moddable_two` it drops into:
> Connecting........_____....._____....._____....._____....._____....._____....._____
> A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header 

If I do the following, it works:
1. Unplug the moddable_two
1. Hold down the flash button (keep it down)
1. Plug in the device
1. While it's building, release the flash button

But, an odd thing, after I get it to build after the steps above, the screen on the device is blank, and the program does not actually seem to run until I disconnect the device and replug it in. I would have expected that the program would immediately run after the build completes and the program is transferred to the device.

I saw issue #250, and tried changing cables, but none made any difference.  I also tried all four ports on the computer.

Any other tips? 
Thanks

andycarle, 2020-05-09:
How very interesting; that's a new one for me. I don't have any pointers at the ready here, but I'm certainly curious to learn more and to try to help out.

Is pressing the reset button after the flash completes enough to get it going again? Is there anything odd about the output from idf.py (when the dance you describe gets it to work)? My deployment output ends like this:

```
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 18368 bytes to 11960...
Wrote 18368 bytes (11960 compressed) at 0x00001000 in 0.2 seconds (effective 820.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 4096.0 kbit/s)...
Hash of data verified.
Compressed 1203408 bytes to 737170...
Wrote 1203408 bytes (737170 compressed) at 0x00010000 in 12.0 seconds (effective 799.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
```

The combination of the two problems you're describing makes me wonder if somehow something is wrong with the reset pin on your programmer or on the Moddable Two itself.

I've also never seen a PC that really needs the device to be unplugged before flashing. Usually just rolling your finger from the reset button to the flash button a couple of times will do it.

mikehedman, 2020-05-09:
@andycarle Thanks for your help.  One of your suggestions fixed one small part - pressing the reset button after the build does initiate running the program.  So that removes one unplug+plug cycle from my dev flow.

My dev flow still feels like I have a big extra step:
1. write code
1. unplug, hold flash button down, plug in, release flash button
1. build happens, ends up by displaying "done", but keeps running (it doesn't return to a prompt)
1. press reset button and program runs
1. modify code
1. ctrl-c to stop the previous run
1. restart at step 2

I'd really like to get rid of step 2 from that flow.

Here's the output from my build, hopefully it'll provide some insight(s) into what's going on:
`$ mcconfig -d -m -p esp32/moddable_two
# xsc main.xsb
# xsl modules
# cc mc.xs.c (slots in flash)
# ld xs_esp.bin
Checking Python dependencies...
Python requirements from /Users/mike/esp32/esp-idf/requirements.txt are satisfied.
Running ninja in directory /Users/mike/dev/moddable/sdk/moddable/build/tmp/esp32/moddable_two/debug/idf
Executing "ninja all"...
[1/8] Performing build step for 'bootloader'
ninja: no work to do.
[6/6] Generating xs_esp32.bin
esptool.py v2.8

Project build complete. To flash, run this command:
../../../../../../../../../../esp32/esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --after hard_reset write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x1000 ../idf/bootloader/bootloader.bin 0x8000 ../idf/partition_table/partition-table.bin 0x10000 ../idf/xs_esp32.bin
or run 'idf.py -p (PORT) flash'
Python requirements from /Users/mike/esp32/esp-idf/requirements.txt are satisfied.
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
esptool.py -p /dev/cu.SLAB_USBtoUART -b 921600 --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 xs_esp32.bin
esptool.py v2.8
Serial port /dev/cu.SLAB_USBtoUART
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:11:ae:70:c5:1c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 18352 bytes to 11959...
Wrote 18352 bytes (11959 compressed) at 0x00001000 in 0.2 seconds (effective 808.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 2557.4 kbit/s)...
Hash of data verified.
Compressed 970896 bytes to 593716...
Wrote 970896 bytes (593716 compressed) at 0x00010000 in 10.0 seconds (effective 780.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Checking Python dependencies...
Running ninja in directory /Users/mike/dev/moddable/sdk/moddable/build/tmp/esp32/moddable_two/debug/idf
Executing "ninja all"...
Running esptool.py in directory /Users/mike/dev/moddable/sdk/moddable/build/tmp/esp32/moddable_two/debug/idf
Executing "/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python /Users/mike/esp32/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/cu.SLAB_USBtoUART -b 921600 --after hard_reset write_flash @flash_project_args"...
Done`

Thanks

phoddie, 2020-05-12:
It should be enough to do this:

1. Hold down the flash button
2. Tap the reset button
3. Release the flash button

When I do that with a serial monitor (115200 baud, 8N1), it boots into programming mode:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

If you reverse the two buttons above, it boots normally:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1012
load:0x40078000,len:10864
load:0x40080400,len:6464
entry 0x400806b8```

mikehedman, 2020-05-14:
@phoddie That works for me. It's a lot easier than messing with unplugging wires.  
Thanks!

closed 2020-05-14 by mikehedman