Moddable-OpenSource/moddable #358: Error with SSL session.alert.level =2 session.alert.description=20

wilberforce, 2020-05-11:
I'm doing a https form post to httpbin.org, and after a while I get this exception:

C:\Users\rhys\Projects\moddable\modules\crypt\ssl\ssl_alert.js (82) # Break: throw!

```
session.alert = {level: s.readChar(), description: s.readChar()};
		if (session.alert.description != close_notify)

82->			throw new Error("SSL: alert: " + session.alert.level + ", " + session.alert.description);
```
Calls stack:
?.unpacketize   /crypt/sslssl_alert:82  https://github.com/Moddable-OpenSource/moddable/blob/public/modules/crypt/ssl/ssl_alert.js#L82
?.unpacketize    /crypt/ssl_record:73
SSLSession.prototype.read:195
SecureSocket.prototype.messageHandler:124

Is this something I should try to catch and ignore or is there something else going on?

```
let request = new Request(requestDictionary);
		request.callback = function (message, value) {
			if (Request.responseComplete === message) {
				trace(value, '\n');
			}
```

phoddie, 2020-05-12:
The SSL alert is sent by the server to the client (ESP32). One reason for that is that detects a problem with the TLS protocol. But... it might not be a client side error. The server could decline a valid connection request for various reasons (rate limiting, for example).

If it happens at all, the SSL alert usual occurs during the initial handshake, which prevents the connection from being established. If you ignore it, the connection is usually terminated by the server anyway. The TLS implementation will catch the exception and report it as a socket close error.

You might also trace the values out of `session.alert.level` and `session.alert.descripton`. Occasionally they offer a hint about what is going on, but the security errors tend to be deliberately opaque.

When you make the request, you can set `trace` in the TLS dictionary. That will show where in the process it is failing, which might provide a clue. To do that:

```javascript
	this.request = new Request({
		...
		Socket: SecureSocket,
		secure: {
			protocolVersion: 0x303,
			trace: true,
		}
```

wilberforce, 2020-05-23:
Thanks

> You might also trace the values out of session.alert.level and session.alert.descripton.
session.alert.level =2 
session.alert.description=20

I'm unclear what these values represent?

I have seen it running on the windows implementation, and it is intermittent in nature. I guess my question is how do I handle this situation in my application?

If ssl_alert.js is throwing the error, and it's async event, how can I catch this?

will a `try/catch` around this catch the negotiation error?

`let request = new Request(requestDictionary);`

closed 2020-08-30 by wilberforce

phoddie, 2020-08-31:
> I guess my question is how do I handle this situation in my application? If ssl_alert.js is throwing the error, and it's async event, how can I catch this?

The `throw` that you reference above in `ssl_alert.js` should be caught by the callback in `SecureSocket` which propagates the error to its client (an HTTP request here) causing the requests's callback to be invoked with the  `Request.error` message. You application can handle that as needed. For example, we've used that in projects to initiate a reconnection.