Moddable-OpenSource/moddable #359: Using "-r 90" doesn't rotate Screen Test

mikehedman, 2020-05-12:
Hi, when I do a build with:
`mcconfig -d -m -p esp32/moddable_two`, the application on the Moddable Two displays in landscape mode as expected.
But if I do:
`mcconfig -d -m -r 90`
the Screen Test simulator displays in portrait mode.
Is there any way to tell Screen Test to use landscape?
I've tried various manifest.json options, such as:
```
	"config": {
		"rotation": 90,
	},
```
But nothing I've tried has had any effect.
Thanks

mikehedman, 2020-05-12:
I just lucked into my own answer.  Closing and restarting Screen Test let the rotation happen as expected.  I noticed that the screen repainted when I would run the `mcconfig` command, but I now see that it doesn't actually reload the new screen...which I guess I would call a bug.

closed 2020-05-12 by mikehedman

phoddie, 2020-05-12:
Thanks for the report. I tested this using the piu/balls example and had a similar impression as you -- something seems broken.. But, balls is not great for testing rotation because it is difficult to visually determine the rotation of the screen contents. Using piu/cards, I found Screen Test works as expected. Let me try to explain:

```
cd $MODDABLE/examples/piu/cards
mcconfig -d -m
```

This runs with the text in the usual orientation. If you haven't changed the defaults, that means it is running in portrait orientation at 240 x 320. Re-run with 90 degree rotation:

```
mcconfig -d -m -r 90
```

You should see that the simulator skin is identical to the previous run (portrait orientation at 240 x 320), but that the content of the display is rotated 90 degrees. Parallel results are shown for rotations of 180 and 270 -- the simulator skin is unchanged and the content rotates within that skin.

You can use the Size menu in Screen Test to change the simulator skin to landscape (320 x 240), portrait orientation (240 x 320), or many others. When you change the Size, a different simulator skin is loaded and the application is relaunched at the new screen size. The rotation of the content within the screen remains whatever you configured at build time.