Moddable-OpenSource/moddable #360: bug? '0123456789abcdef'[175 % 16] === undefined

wmhilton, 2020-05-12:
I noticed that I kept getting SHA1s that contained the word `undefined` in them and well... this is the origin. Occasionally, on macOS, `chars[byte % 16]` is returning undefined.

See value of x in Locals pane:
[image]

If you plug in the values for `chars` and `byte` you should get `'f'`.

It's fairly repeatable... can I produce some kind of debugging data to help you like a heap snapshot or something?

phoddie, 2020-05-12:
I'm not able to reproduce this. I made two attempts, both on macOS and ESP32. Both ran by replacing the main.js of the helloworld example. Here's the first test: based on the title of the bug report

```javascript
while (true) {
	let foo = '0123456789abcdef'[175 % 16];
	trace(foo, "\n");
	if ("f" !== foo)
		debugger;
}
```

Here's the second based on the screen capture provided:

```javascript
while (true) {
	let hex = "";
	const chars = "0123456789abcdef";
	let bytes = new Uint8Array(1024);
	for (let i = 0; i < bytes.length; i++)
		bytes[i] = Math.random() * 255;
	let buffer = bytes.buffer;
	for (const byte of new Uint8Array(buffer)) {
		const i = chars[byte >> 4];
		if (i === undefined) debugger;
		hex += i;
		const x = chars[byte % 16];
		if (x === undefined) debugger;
		hex += x;
	}
	trace(hex, "\n");
}
```

wmhilton, 2020-05-12:
Well, I mean obviously it can't be that simple. That's why I'm not reporting it as a bug as much as asking for tips how to dig deeper. My best guesses at this point are some kind of memory overflow / pointer corruption, but I'm not sure how to see the memory usage. XS precompiled code right? Where does it save the generated bytecode, maybe that would show something.

wmhilton, 2020-05-13:
OK so.... good and bad news. None of the issues I was having yesterday are reproducible today, and the only thing that's different is I had to reboot my computer this morning. So... total wash. Ye old turn it off and on again mysteriously fixes all errors. 🙏 

But good news is it's all working now, which is rather exciting. Will send email.

I'm gonna blame this bug on Apple. Maybe one day it will rear its head again, but it does no good to keep the issue open... closed issues are still indexed by Google so 🤷‍♂️

closed 2020-05-13 by wmhilton