Moddable-OpenSource/moddable #365: Cannot create proper font for Indian fonts like Devnagiri

arahasya, 2020-05-20:
Hi

I am trying to create font according to the documentation

In preview I am using the word शिष्टाचार
Here is the SS. It is not getting rendered properly. 

[image]

lprader, 2020-05-20:
Hello! I haven't used the preview feature of Glyph Designer, so I'm not sure what the problem is there. But if you follow the instructions in the documentation to export the font and try to render that same string in an application built with the Moddable SDK, does it work as expected?

arahasya, 2020-05-21:
I am having a hard time finding an open source font generator 

Can you test adding a devnagiri font?

arahasya, 2020-05-21:
okay I got the license of Glyph Designer and copied the font and png to the assets

I am get this message 

### Error: Invalid BMF header

arahasya, 2020-05-21:
As I wsa afraid Devnagiri script is not getting rendered properly

word - शिष्टाचार

Output -

[image]

How to try this with monofont which claims to support Indian languages?

phoddie, 2020-05-22:
I'm not familiar with Devanagari, but the Wikipedia page suggests there are contextual forms. If that is the case, it requires a shaper to render properly. We have successfully contextual forms with Arabic and Thai using Monotype's WorldType Shaper with their Spark text engine.

If you'd like to try that, you'll need to get in touch with Monotype. They license their technology separately. If you would like us to make an introduction to begin that conversation, send an e-mail to info@moddable.com.

phoddie, 2020-08-04:
Closing due to inactivity.

closed 2020-08-04 by phoddie