Moddable-OpenSource/moddable #366: Cannot see any text in Hello World Example

arahasya, 2020-05-21:
Hi

I have build on macOS 
when I run mcconfig -d -m I can see the display but there is not text

[image]

What may be the issue?

arahasya, 2020-05-21:
This is weird because I can see text with other examples

[image]

lprader, 2020-05-21:
If you look at the source code, the `helloworld` example does not display text on the screen. It breaks into the debugger and traces the message "Hello, world - sample" to the console.

closed 2020-05-21 by arahasya