Moddable-OpenSource/moddable #368: ESP32  Preference.keys not implemented

wilberforce, 2020-05-25:
Had me going for a while until I saw it's returning an empty array.

Perhaps a warning should be output?

https://github.com/Moddable-OpenSource/moddable/blob/public/modules/files/preference/esp32/modPreference.c#L149

```
void xs_preference_keys(xsMachine *the)
{
	xsResult = xsNewArray(0);
}
```

I started to write the code - then found the function `nvs_entry_find` needed are in version 4 of the idf. Here is the starting point for the future when get to that version:

```
void xs_preference_keys(xsMachine *the)
{
	xsResult = xsNewArray(0);
	const char * domain = xsmcToString(xsArg(0));
	nvs_iterator_t it = nvs_entry_find("nvs", domain, NVS_TYPE_ANY);
	while (it != NULL) {
			nvs_entry_info_t info;
			nvs_entry_info(it, &info);
			it = nvs_entry_next(it);
			//printf("key '%s', type '%d' \n", info.key, info.type);
			xsVar(0) = xsString(info.key);
			xsCall1(xsResult, xsID_push, xsVar(0));
	}
}
```

phoddie, 2020-05-27:
Ha. Yes, I recently stumbled upon `nvs_entry_find` but found it required ESP-iDF v4. When we do update to that, we'll add support. Meanwhile, perhaps adding an `xsTrace` to `xs_preference_keys` would be enough?

andycarle, 2021-01-29:
This issue's time has finally come! I've enabled `Preference.keys` for esp32 and it is working well. 

That will be available in our next open source push.

wilberforce, 2021-01-29:
I was thinking about looking at this... thanks for that.

When does the idf-4 branch go mainstream?

andycarle, 2021-01-29:
The announced plan is "early February." Unless something big comes up, I think we're on pace to hit that target. 

@wilberforce ESP-IDF 4.2 builds are working well for you? I'm personally very happy with the builds and (especially) the installation process for 4.2.

andycarle, 2021-02-03:
This change is now live. @wilberforce , please give it a shot when you can and close this out if everything looks good to you.

wilberforce, 2021-02-04:
I'll give it a try.

wilberforce, 2021-02-09:
Confirmed - preferences can now be read on the ESP32

closed 2021-02-09 by wilberforce