Moddable-OpenSource/moddable #369: Update needed to Keeping Up To Date.md for macOS Host env. update

mikew67, 2020-05-25:
In https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Keeping%20Up%20To%20Date.md#host-mac (step 3)
 the "make" command may fail due to the user's Xcode configuration:

- If the user has "Xcode beta" installed, the make command will fail because it's looking for "Xcode" explicitly.
- Probably will also fail if the user hasn't gone through the "Install Command Lines" post-Xcode-installation step.
- Creating an alias to the Xcode beta called "Xcode.app" doesn't suffice.  Creating a hard link probably would. But that's not normally a Mac thing and the user might not know how to do it.
- Renaming Xcode by dropping the " beta" works for this issue (but, IIRC, may cause other issues later)

**Suggest you add a Caution note to the doc**, with a suggestion to rename Xcode, etc., to fix the problem.

**Editorial Note:**  Many thanks for the Keeping Up to Date procedure and doc.

mikew67, 2020-05-25:
Oh, and in (Step 1), the "git pull" will undoubtedly fail because the user has made mods to the repo.  Add a note to have them stash their mods before the pull.

phoddie, 2020-05-27:
@andycarle - I'm not sure about supporting Xcode beta releases, but @mikew67's suggestion to add a note could help some other people.

andycarle, 2020-05-27:
I'll add a note about Xcode versions to the Keeping Up to Date doc. 

On the "git pull" it's true that if one has edited files in the repo that have been subsequently updated remotely, you'll need to stash and merge. I can add a note to that effect, but the specific steps for doing the merge in the general case are probably beyond the scope of this document.

mikew67, 2020-05-27:
Just a suggestion to do a stash would be great.  Would help git newbies unblocked.  Beyond that they can check out git docs,

andycarle, 2020-06-06:
This update to the documentation has been pushed to GitHub.

Please don't hesitate to let me know if you have any further suggestions!

closed 2020-06-06 by andycarle