Moddable-OpenSource/moddable #370: Update docs to indicate which git errors are significant

mikew67, 2020-05-25:
While using the Keeping Up to Date doc/ESP32 Update section, various git errors or warnings may be encountered, e.g., 
`% git checkout v3.3.2 
error: The following untracked working tree files would be overwritten by checkout:
	examples/system/ota/OTA_workflow.png
Please move or remove them before you switch branches.
Aborting
`
` % git checkout v3.3.2
warning: unable to rmdir 'components/bootloader/subproject/components/micro-ecc/micro-ecc': Directory not empty
warning: unable to rmdir 'components/bt/controller/lib': Directory not empty
warning: unable to rmdir 'components/bt/host/nimble/nimble': Directory not empty
warning: unable to rmdir 'components/esp_wifi/lib_esp32': Directory not empty
Updating files: 100% (4424/4424), done.
M	components/coap/libcoap
M	components/expat/expat
M	components/lwip/lwip
M	components/mqtt/esp-mqtt
Previous HEAD position was 463a9d8b7 Merge branch 'bugfix/ci_deploy_tags_v4.0' into 'release/v4.0'
HEAD is now at 9e70825d1 Merge branch 'bugfix/sntp_init_can_run_before_net_connection_v3.3' into 'release/v3.3'
 %`

phoddie, 2020-05-27:
@andycarle - can you help?

andycarle, 2020-05-27:
Hi @mikew67, good to hear from you. 

Thank you for pointing this out. What you're seeing here are the submodules in the esp-idf repository misbehaving a bit. Basically, git can get confused when switching between very different branches that contain conflicting submodules, especially when going "back in time" (in your case, from v4.0 to v3.3.2). If you're curious, you can learn more in the "[Issues with Submodules](https://git-scm.com/book/en/v2/Git-Tools-Submodules#Issues-with-Submodules)" section of the [Pro Git book](https://git-scm.com/book/en/v2).

Bottom line, it would be better to recommend that people use the `--recurse-submodules` flag when checking out the v3.3.2 tag. This flag basically says, "do the checkout and the submodule update as one step," which avoids the inconsistent middle state of doing it in two commands. That would eliminate these errors and warnings when switching to v.3.3.2 from a wider variety of initial git states. 

I'll get updated instructions into the guide. But, in the meantime, this is the better way to go:

```shell
cd ~/esp32/esp-idf
git fetch
git checkout --recurse-submodules v3.3.2
```
&nbsp;- Andy

andycarle, 2020-06-06:
This update to the instructions has been pushed to GitHub.

Please let me know if you have any further difficulties!

closed 2020-06-06 by andycarle