Moddable-OpenSource/moddable #371: Gecko support is broken

mikew67, 2020-05-25:
Has Gecko support been dropped?

The build instructions the following location don't work anymore:
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/gecko/GeckoBuild.md

In particular, the following file and path no longer exist:
$MODDABLE/tools/mcconfig/gecko/make.subplatform.mk

mkellner, 2020-05-26:
Can you please provide the build command you're using? `mcconfig ...`

mikew67, 2020-05-26:
Re question from Kellner -- never got that far.  See and try to follow .md file build instructions.

mkellner, 2020-05-26:
I'm sorry. I don't have that board. I thought I might have an easy response to unblock you.

If there is no subplatform defined for the board you're using, it's not likely to work. Since you've suggested that it is similar to the Thunderboard 2, perhaps you can look at the  thunderboard2 port for clues on getting it going? There are specifics relative to the board in the build/devices/gecko/targets/thunderboard2 (etc.) files.

If you look at `build/devices/gecko/targets/thunderboard2/make.thunderboard2.mk`, you'll see the GECKO_BOARD and GECKO_PART defined, among other things.

As to the mcconfig request above:
The `gecko` platform needs a `subplatform` defined. We've done four of them (as described in the doc). If one tries to build with `mcconfig -d -m -p gecko` it will fail. The command needs to be something along the lines of `mcconfig -d -m -p gecko/thunderboard2`.

mikew67, 2020-05-27:
Michael K, thanks for the porting hints.

Now that I look at your [SDK improvements](https://blog.moddable.com/blog/moddable-sdk-improvements-for-esp32-projects/), I see that they're limited to the ESP32 platform. (And I note that all the Gecko support is missing from this version.)

Does that mean that I need to keep the older SDK around for Gecko support?  Or do you plan to upgrade it and merge into the new SDK anytime soon?

Thanks,

Mike

PS: And I note that it's bit difficult to keep two versions of the SDK, etc., around because of magic pathnames, etc.

mikew67, 2020-05-27:
Whoops, didn't mean to close this yet.