Moddable-OpenSource/moddable #373: Monitor module is broken

bartmichu, 2020-06-01:
**Build environment:** Linux
**Target device:** NodeMCU ESP8266

**Description**
I'm getting errors when building code with the Monitor module.
```
...
# ld main.bin
/home/redacted/moddable/build/tmp/esp/debug/toggle-led/mc.xs.c.o:(.irom0.str.1+0x30): undefined reference to `xs_digital_monitor_get_pwm_duty'
/home/redacted/moddable/build/tmp/esp/debug/toggle-led/mc.xs.c.o:(.irom0.str.1+0x38): undefined reference to `xs_digital_monitor_get_pwm_freq'
/home/redacted/moddable/build/tmp/esp/debug/toggle-led/mc.xs.c.o:(.irom0.str.1+0x7fa8): undefined reference to `xs_digital_monitor_get_pwm_freq'
/home/redacted/moddable/build/tmp/esp/debug/toggle-led/mc.xs.c.o:(.irom0.str.1+0x7fd8): undefined reference to `xs_digital_monitor_get_pwm_duty'
collect2: error: ld returned 1 exit status
make: *** [/home/redacted/moddable/build/tmp/esp/debug/toggle-led/makefile:478: /home/redacted/moddable/build/bin/esp/debug/toggle-led/main.bin] Error 1
The terminal process terminated with exit code: 2
```

**Steps to Reproduce**
Try to compile this code:
```
import Digital from 'pins/digital';
import Monitor from 'pins/digital/monitor';

const led = new Digital({ pin: 2, mode: Digital.Output });

const button = new Digital({
  pin: 0,
  mode: Digital.Input,
});

const buttonMonitor = new Monitor({
  pin: button.pin,
  mode: Digital.Input,
  edge: Monitor.Falling,
});

buttonMonitor.onChanged = function onChanged() {
  led.write(!led.read());
};
```

**Other information**
This code was working fine about two months ago, so the latest change seems to be the culprit ;)

mkellner, 2020-06-01:
My apologies. It will be fixed in the next OpenSource drop.
In the meantime, you can add these lines to the bottom of `modules/pins/digital/monitor/esp/modMonitor.c`

```
void xs_digital_monitor_get_pwm_duty(xsMachine *the) {}
void xs_digital_monitor_get_pwm_freq(xsMachine *the) {}
```

bartmichu, 2020-06-01:
Great, thank you for your help and your work on this.

closed 2020-08-04 by phoddie