Moddable-OpenSource/moddable #381: "Resource busy" on USB port

markingle, 2020-06-09:
**Build environment:** macOS Catalina 10.15.4
**Target device:** Moddable Two, 

**Description**
When trying to upload the hello world example the flash fails due to a Resource Busy error.  There are no other applications running and connected to the USB port.

**Steps to Reproduce**
Run "mcconfig -d -m -p esp32"

**Expected behavior**
Hello World to run on the MOddable

**Images**
[image]

andycarle, 2020-06-09:
Hi @markingle,

I suspect that what is happening here is that Espressif's idf.py tool is picking the wrong (virtual) serial port on your computer to attempt to upload to. It is choosing `/dev/cu.usbserial-0001`, but the Moddable programmer typically shows up in macOS as `/dev/cu.SLAB_USBtoUART` if the proper Silicon Labs VCP driver has been installed.

I recommend trying out the instructions in the [Getting Started guide's macOS ESP32 setup section](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#esp32-mac), step number 9. There is a note there about setting the optional `UPLOAD_PORT` environment variable and how to determine the proper port to use. 

Please give that a try and let me know how it goes. If that's not the issue, we'll keep digging.

Cheers!
&nbsp;- Andy

markingle, 2020-06-09:
Yep! That got it!  Thanks for the quick reply!

closed 2020-06-09 by markingle