Moddable-OpenSource/moddable #383: error invalid character from regexps from babel-standalone

dckc, 2020-06-12:
**Build environment:** Linux
**Target device:** linux

**Description**
I'm investigating the use of `@agoric/babel-standalone` in XS for module loading (ref https://github.com/Agoric/SES-shim/pull/336 https://github.com/Agoric/agoric-sdk/issues/57 ) but I get `invalid character` errors from regexps used there.

```
~/projects/moddable/examples/babel-standalone$ cat -n node_modules/@agoric/babel-standalone/babel.js | egrep '21342|66549|68362'
 21342	    return JSON.parse(str.replace(/^\)]}'[^\n]*\n/, ''));
 66549	        } else if (!hasUnicodeFlag && (res = matchReg(/^(?:]|})/))) {
 68362	      if (hasFeature$1(features, FEATURES$1.unicodePropertyEscape) && /\\[pP]{/.test(pattern)) {
```

cc @michaelfig @kriskowal @warner

**Steps to Reproduce**

```
~/projects/moddable/examples/babel-standalone$ npm install @agoric/babel-standalone
...
+ @agoric/babel-standalone@7.9.5
~/projects/moddable/examples/babel-standalone$ grep babel package-lock.json 
    "@agoric/babel-standalone": {
      "resolved": "https://registry.npmjs.org/@agoric/babel-standalone/-/babel-standalone-7.9.5.tgz",
~/projects/moddable/examples/babel-standalone$ wc node_modules/@agoric/babel-standalone/babel.js 
 100608  269495 3035064 node_modules/@agoric/babel-standalone/babel.js

~/projects/moddable/examples/babel-standalone$ ls
build  main.js  manifest.json  node_modules  package-lock.json

~/projects/moddable/examples/babel-standalone$ cat main.js 
trace("hello\n");

~/projects/moddable/examples/babel-standalone$ cat manifest.json 
{
  "include": "$(MODDABLE)/examples/manifest_base.json",
  "modules": {
    "*": [ "./main" ],
    "babel": "node_modules/@agoric/babel-standalone/babel"
  },
}

~/projects/moddable/examples/babel-standalone$ mkdir -p build && mcconfig -o build -d -m
# xsc babel.xsb
# mcrez resources
/home/connolly/projects/moddable/examples/babel-standalone/node_modules/@agoric/babel-standalone/babel.js:21342: error: ^\)] invalid character!
/home/connolly/projects/moddable/examples/babel-standalone/node_modules/@agoric/babel-standalone/babel.js:66549: error: ^(?:] invalid character!
/home/connolly/projects/moddable/examples/babel-standalone/node_modules/@agoric/babel-standalone/babel.js:68362: error: \\[pP]{ invalid quantifier!
Total resource size: 0 bytes
### 3 error(s)
# cc mc.resources.c
/home/connolly/projects/moddable/examples/babel-standalone/build/tmp/lin/debug/mc/babel-standalone/makefile:288: recipe for target '/home/connolly/projects/moddable/examples/babel-standalone/build/tmp/lin/debug/mc/babel-standalone/modules/babel.xsb' failed
make: *** [/home/connolly/projects/moddable/examples/babel-standalone/build/tmp/lin/debug/mc/babel-standalone/modules/babel.xsb] Error 22
make: *** Waiting for unfinished jobs....

~/projects/moddable/examples/babel-standalone$ cat -n node_modules/@agoric/babel-standalone/babel.js | egrep '21342|66549|68362'
 21342	    return JSON.parse(str.replace(/^\)]}'[^\n]*\n/, ''));
 66549	        } else if (!hasUnicodeFlag && (res = matchReg(/^(?:]|})/))) {
 68362	      if (hasFeature$1(features, FEATURES$1.unicodePropertyEscape) && /\\[pP]{/.test(pattern)) {
```

**Expected behavior**
babel-standalone is compiled

**Other information**
I suspect this is related to one or more of the known regexp test failures in [XS Conformance](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20Conformance.md)

dckc, 2020-07-03:
update: https://github.com/agoric-labs/babel/issues/4#issuecomment-653205125

phoddie, 2020-07-08:
These RegExps depend on Annex B (JavaScript for web browsers) features. Strictly speaking, an engine is not required to support them.

We've implemented support for this particular case. That will be included in our next Moddable SDK push.

We do not intend to provide general Annex B support.

This issue should be reported to the library authors so they can modify their RegExps to eliminate the unnecessary dependency. They should be simple changes.

phoddie, 2020-08-04:
Resolved.

closed 2020-08-04 by phoddie