Moddable-OpenSource/moddable #384: Enable to execute example on esp8266

rgujju, 2020-06-13:
Hello,

I am trying to run the hello word example on the esp8266 but i am not able to do so. The example gets built and flashed but doesnt seem to execute. 
I use the command `mcconfig -d -m -p esp` , and the xsbug window launches but there are no buttons for executing, stepping, etc as shown in the image below. I am able to execute the helloworld example in the simulator using `mcconfig -d -m -p lin`. 

What could I be missing?

[image]

andycarle, 2020-06-16:
Hi @rgujju. I'm sorry to hear that you're having trouble getting xsbug to connect to the app running on your ESP8266. Let's see if we can sort out what's going on here.

It looks from your image like the app probably is flashing and perhaps is running successfully. But that it is then failing to connect to xsbug for interactive debugging. 
 
Can you answer a few questions for me to help you narrow down the potential causes here?

1) What sort of ESP8266 development device are you flashing to? A NodeMCU board or a Moddable One or some other development kit?

2) If the device does not have a screen but does have a built-in LED, can you try running the [blink example](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/pins/blink/main.js) to blink the built-in LED? You may need to change the pin number in the example to get it to work with your board if you're not using a NodeMCU board.

3) Sometimes simple fixes can help with flashing and debugging problems. You might try a different USB cable or a different USB port on your computer.

Hopefully, we can narrow down the issue quickly.

Cheers,
&nbsp;- Andy

rgujju, 2020-06-16:
hello @andycarle 
1. I am using a NodeMCU board.
2. I have tried the blink example as well and the LED does not blink.
3. I have tried a different USB cable and port but still does not work.

bartmichu, 2020-06-16:
Hi @rgujju . In my (limited) experience it always comes down to the serial converter. I've tried these ESP8266 boards:
- WEMOS D1 mini Pro with CH340C
- NodeMCU V3 with CH340G
- NodeMCU V1.0 with CP2102
- Moddable One with CP2102

Only CP2102 boards are working with default settings for me. For everything else I have to tweak ```UPLOAD_SPEED``` and ```DEBUGGER_SPEED```.

You could start with:
```
export UPLOAD_SPEED=115200 DEBUGGER_SPEED=115200 && mcconfig -d -m -p esp
```

andycarle, 2020-06-17:
@rgujju The suggestions made above by @bartmichu are reasonable and definitely worth a shot. I haven't had any issues with the default speeds when using the NodeMCU board, but results can definitely vary from computer to computer and USB port to USB port.

Try bumping down the upload and debugger speeds and see if that does you any good. If it does not, could you try setting the VERBOSE environment variable to 1 and the building again, then could you post the build log?

Thank you!
&nbsp;- Andy

rgujju, 2020-06-17:
Hi @andycarle ,
The method mentioned by @bartmichu  does not work for me. Tried with 9600 as well. 
Here is the verbose log

```
VERBOSE=1 mcconfig -d -m -p esp

/home/rgujju/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py -b 921600 -p /dev/ttyUSB0 write_flash --flash_freq 80m --flash_mode qio --flash_size 4MB 0x0000 /home/rgujju/esp/ESP8266_RTOS_SDK/components/esp8266/firmware/boot_v1.7.bin 0x1000 /home/rgujju/Documents/Embedded/frameworks/moddable/build/bin/esp/debug/blink/main.bin 0x3FC000 /home/rgujju/esp/ESP8266_RTOS_SDK/components/esp8266/firmware/esp_init_data_default.bin
esptool.py v2.4.0
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:b2:f4:57
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash params set to 0x004f
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.0 seconds (effective 933.4 kbit/s)...
Hash of data verified.
Compressed 532804 bytes to 325963...
Wrote 532804 bytes (325963 compressed) at 0x00001000 in 5.1 seconds (effective 836.3 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 547.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/home/rgujju/Documents/Embedded/frameworks/moddable/build/bin/lin/debug/serial2xsbug /dev/ttyUSB0 921600 8N1
```

andycarle, 2020-06-17:
Very strange. Your flashing output looks about the same as mine does when I run on Ubuntu:

```
acarle@DESKTOP-MS3D381:~/Projects/moddable/examples/helloworld$ VERBOSE=1 mcconfig -d -m -p esp
/home/acarle/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py -b 921600 -p /dev/ttyS3 write_flash --flash_freq 80m --flash_mode qio --flash_size 4MB 0x0000 /home/acarle/esp/ESP8266_RTOS_SDK/components/esp8266/firmware/boot_v1.7.bin 0x1000 /home/acarle/Projects/moddable/build/bin/esp/debug/helloworld/main.bin 0x3FC000 /home/acarle/esp/ESP8266_RTOS_SDK/components/esp8266/firmware/esp_init_data_default.bin
esptool.py v2.4.0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: cc:50:e3:55:f5:33
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash params set to 0x004f
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.0 seconds (effective 959.7 kbit/s)...
Hash of data verified.
Compressed 438580 bytes to 276336...
Wrote 438580 bytes (276336 compressed) at 0x00001000 in 4.0 seconds (effective 887.4 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 515.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/home/acarle/Projects/moddable/build/bin/lin/debug/serial2xsbug /dev/ttyS3 921600 8N1
```
I'd like to try one more thing to narrow down whether this is an issue with your build for the ESP8266 (i.e. the build output is flashing correctly but failing to run because it is somehow incorrect) or just a problem with the connection to xsbug.

Could you try running the [blink example](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/pins/blink/main.js) again, but this time do a release build? So the commands would be:

```shell
cd $MODDABLE/examples/pins/blink
mcconfig -m -p esp
```

If that gets the LEDs on the board blinking, then we'll know for sure that this is an issue with connecting to xsbug and we can work from there.

Thanks!
&nbsp;- Andy
```

rgujju, 2020-06-18:
Still does not work. When I type in the command it directly uploads, is there a way to build again and then upload?

```
VERBOSE=1 mcconfig -m -p esp
/home/rgujju/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py -b 921600 -p /dev/ttyUSB0 write_flash --flash_freq 80m --flash_mode qio --flash_size 4MB 0x0000 /home/rgujju/esp/ESP8266_RTOS_SDK/components/esp8266/firmware/boot_v1.7.bin 0x1000 /home/rgujju/Documents/Embedded/frameworks/moddable/build/bin/esp/release/blink/main.bin 0x3FC000 /home/rgujju/esp/ESP8266_RTOS_SDK/components/esp8266/firmware/esp_init_data_default.bin
esptool.py v2.4.0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:b2:f4:57
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash params set to 0x004f
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.0 seconds (effective 931.3 kbit/s)...
Hash of data verified.
Compressed 411140 bytes to 260170...
Wrote 411140 bytes (260170 compressed) at 0x00001000 in 4.0 seconds (effective 815.1 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 343.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

wilberforce, 2020-06-18:
Looking at the log - thr baud rate has not changed ... It's still at the default speeds

rgujju, 2020-06-18:
Hi @wilberforce 
Changing the baud rate does not make a difference on my setup. Here's the log with 9600 baud rate.
```
UPLOAD_SPEED=9600 DEBUGGER_SPEED=9600 VERBOSE=1 mcconfig -d -m -p esp
/home/rgujju/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py -b 9600 -p /dev/ttyUSB0 write_flash --flash_freq 80m --flash_mode qio --flash_size 4MB 0x0000 /home/rgujju/esp/ESP8266_RTOS_SDK/components/esp8266/firmware/boot_v1.7.bin 0x1000 /home/rgujju/Documents/Embedded/frameworks/moddable/build/bin/esp/debug/blink/main.bin 0x3FC000 /home/rgujju/esp/ESP8266_RTOS_SDK/components/esp8266/firmware/esp_init_data_default.bin
esptool.py v2.4.0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:b2:f4:57
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x004f
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 3.1 seconds (effective 10.5 kbit/s)...
Hash of data verified.
Compressed 447892 bytes to 280762...
Wrote 447892 bytes (280762 compressed) at 0x00001000 in 295.2 seconds (effective 12.1 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.1 seconds (effective 8.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/home/rgujju/Documents/Embedded/frameworks/moddable/build/bin/lin/debug/serial2xsbug /dev/ttyUSB0 9600 8N1
```

andycarle, 2020-06-19:
Hi @rgujju,

I've been kicking around ideas with the rest of the team and we have a few suggestions to try. 

1) This is **very** unlikely to be the same problem you are experiencing, but I happened to encounter similar symptoms while I was working today. In my case, the issue was that I had two copies of xsbug open: one in Windows and one in my Ubuntu VM. I had to close both and reopen xsbug in Windows to get the port to bind correctly and the connection to the ESP8266 working. You might also make sure you don't have a zombie xsbug process running somehow by checking `ps -u`.

2) A possibility is that something has become corrupted in flash on your device. You could try completely erasing the ESP8266 flash and then flashing again. To do this on your system, run:

```shell
/home/rgujju/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py -b 115200 -p /dev/ttyUSB0 erase_flash
```
Then try your build/deploy/run again.

3) Another possibility is that the flash storage module on your ESP8266 dev board cannot run at the frequency that the Moddable SDK tries by default. To test this, please try replacing [$MODDABLE/build/devices/esp/manifest.json](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp/manifest.json) with [this version](https://github.com/Moddable-OpenSource/moddable/files/4801796/manifest.txt) (rename to manifest.json; apparently GitHub does not allow .json files to be attached). 

4) If none of the above ideas work, you could try flashing a pre-built binary to your device to see if it runs. I have built a release build of the blink app and put the binary outputs in this zip:
[blink.zip](https://github.com/Moddable-OpenSource/moddable/files/4801799/blink.zip).

Please extract them to a directory and then run the following commands:

```shell
/home/rgujju/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py -b 115200 -p /dev/ttyUSB0 erase_flash
/home/rgujju/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py -b 115200 -p /dev/ttyUSB0 write_flash --flash_freq 40m --flash_mode qio --flash_size 4MB 0x0000 ./boot_v1.7.bin 0x1000 ./main.bin 0x3FC000 ./esp_init_data_default.bin
```
If that gets the LEDs on your dev board blinking, then we would think the issue is with your build and not the deployment step.

Cheers!
 - Andy

rgujju, 2020-06-20:
Following are the results
1) There is no other xsbug process running.
2) Even erasing the flash and then deploying does not work
3) Does not work either, here are the logs
```
UPLOAD_SPEED=9600 DEBUGGER_SPEED=9600 VERBOSE=1 mcconfig -d -m -p esp
# xsl modules
/home/rgujju/Documents/Embedded/frameworks/moddable/build/bin/lin/release/xsl -b /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modules -o /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink -p time.xsb -p timer.xsb -p Resource.xsb -p instrumentation.xsb -p pins/digital.xsb -p mc/config.xsb -s "*" -c 1536,512,512,64,256,32,53,3,1024,17,32768,main /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modules/time.xsb /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modules/timer.xsb /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modules/Resource.xsb /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modules/instrumentation.xsb /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modules/main.xsb /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modules/pins/digital.xsb /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modules/mc/config.xsb /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modTime.c.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modTimer.c.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/timer.c.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/Resource.c.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modInstrumentation.c.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modPreference.c.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modLwipSafe.c.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modGPIO.c.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/digital.c.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modTimer.h.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modInstrumentation.h.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modLwipSafe.h.xsi /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/modGPIO.h.xsi
# mcrez resources
/home/rgujju/Documents/Embedded/frameworks/moddable/build/bin/lin/release/mcrez   -o /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink -p esp -r mc.resources.c
Total resource size: 0 bytes
# cc mc.resources.c (slots in flash)
/home/rgujju/esp/toolchain/Linux/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/mc.resources.c -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266  -DmxUseDefaultSharedChunks=1 -DmxRun=1 -DmxNoConsole=1 -DkCommodettoBitmapFormat=7 -DkPocoRotation=0 -DmxDebug=1 -DDEBUGGER_SPEED=9600 -DMODINSTRUMENTATION=1 -DmxInstrument=1 -DkESP8266Version=23 -I/home/rgujju/Documents/Embedded/frameworks/moddable/modules/base/timer -I/home/rgujju/Documents/Embedded/frameworks/moddable/modules/base/instrumentation -I/home/rgujju/Documents/Embedded/frameworks/moddable/modules/network/socket/lwip -I/home/rgujju/Documents/Embedded/frameworks/moddable/modules/pins/digital/esp -I/home/rgujju/esp/esp8266-2.3.0/tools/sdk/include -I/home/rgujju/esp/esp8266-2.3.0/tools/sdk/lwip/include -I/home/rgujju/esp/esp8266-2.3.0/cores/esp8266 -I/home/rgujju/esp/esp8266-2.3.0/variants/generic -I/home/rgujju/esp/esp8266-2.3.0/cores/esp8266/spiffs -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/devices/esp/lib/tinyi2s/ -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/devices/esp/lib/bsearch/ -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/devices/esp/lib/fmod/ -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/devices/esp/lib/rtc/ -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/devices/esp/lib/tinyi2s/ -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/devices/esp/lib/tinyprintf/ -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/devices/esp/lib/tinyuart/ -I/home/rgujju/esp/esp8266-2.3.0/cores/esp8266/ -I/home/rgujju/esp/esp8266-2.3.0/cores/esp8266/spiffs/ -I/home/rgujju/esp/esp8266-2.3.0/cores/esp8266/umm_malloc/ -I/home/rgujju/Documents/Embedded/frameworks/moddable/xs/includes -I/home/rgujju/Documents/Embedded/frameworks/moddable/xs/sources -I/home/rgujju/Documents/Embedded/frameworks/moddable/xs/sources/pcre -I/home/rgujju/Documents/Embedded/frameworks/moddable/xs/platforms/esp -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/devices/esp -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/lib -I/home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -o /home/rgujju/Documents/Embedded/frameworks/moddable/build/tmp/esp/debug/blink/mc.resources.c.o.unmapped
[... cut ...]
Wrote 447892 bytes (280762 compressed) at 0x00001000 in 295.1 seconds (effective 12.1 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.1 seconds (effective 8.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/home/rgujju/Documents/Embedded/frameworks/moddable/build/bin/lin/debug/serial2xsbug /dev/ttyUSB0 9600 8N1
```

4) Does not work either

If it is of any help, These are options i use on arduino:
[image]

andycarle, 2020-06-26:
Hi @rgujju, I'm sorry to hear that the issues continue. 

I'm running out of good ideas for remote debugging here. But I would love to get my hands on the device that is causing you all of these problems to see if I can figure it out. 

Would you be interested in a swap? We could mail you a known-working NodeMCU board and you could mail us back the one causing issues for you here. Just send me an email ( andy@moddable.com ) if you're interested!

Cheers,
&nbsp;- Andy

mrpeu, 2020-07-02:
Hi @rgujju , I've only read quickly through this thread so pardon me if that's already been explored.
Have you checked if you're using the correct voltage for the upload? I've had the same upload behavior when [I was using 5V instead of 3 on a Sonoff S20](https://github.com/Moddable-OpenSource/moddable/issues/346#issuecomment-653150207). 🤷‍♀️

burtonkent, 2020-08-15:
Hi, I'm running into the same issues as rgujju. It gets stuck on "Hard resetting via RTS pin..." when trying to use the debugger. 

I did directly write the blink program (from blink.zip, above) to /dev/cu.SLAB_USBtoUART on a Mac, it uploaded fine, did the "Hard resetting via RTS pin..." and went back to the command line. The LEDs aren't blinking though. In case I was supposed to activate something or other, I tried pressing the buttons on the NodeMCU, also disconnecting/reconnecting the USB power/data line. Sorry if I missed the how to activate in the documentation.

I do not get this line in my output:

` /home/rgujju/Documents/Embedded/frameworks/moddable/build/bin/lin/debug/serial2xsbug /dev/ttyUSB0 9600 8N1 `

In case that's a command I need to run, I tried:
`/build/bin/lin/debug/serial2xsbug /dev/cu.SLAB_USBtoUART 9600 8N1`

And it said no such file or directory. I need to head over to a friend's house, I'm late. I'll come back and review the documentation later to find out if I missed something. Would be happy to mail you my NodeMCU, I have 3. No swap needed. [This ebay listing is where I bought them from](https://www.ebay.com/itm/ESP32-ESP-32S-NodeMCU-Development-Board-2-4GHz-WiFi-Bluetooth-Dual-Mode-CP2102/223144268292).

phoddie, 2020-08-15:
I'm late to this adventure, but I'm curious..  @burtonkent or @rgujju, after you install the app with `mcconfig` could you try running a serial monitor:

- For ESP8266 at 74880 baud (8 bit, no parity, 1 stop bit).
- For ESP32, at 115200 baud (8 bit, no parity, 1 stop bit).

Reboot the device. I'd like to know what is received after `ets Jan  8 2013` or `ets Jun  8 2016`. Thank you.

(I've ordered the NodeMCU board referenced above.)

burtonkent, 2020-08-16:
@phoddie I'll do that this week. I'm also going to try installing on Windows and/or Linux to see if it's the OS or the board that's the issue.

phoddie, 2020-08-25:
@burtonkent - I received the ESP32 board referenced above. I was able to run $MODDABLE/examples/helloworld without a problem from macOS.

```
mcconfig -d -m -p esp32`
```

Some additional information is necessary to help diagnose the issue.

rgujju, 2020-08-25:
Hello @phoddie 

This is the output i get.

```
` �9K
load 0x40100000, len 2592, room 16 
tail 0
chksum 0xef
load 0x00000000, len 0, room 8 
tail 0
chksum 0xef
load 0x00000000, len 0, room 0 
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c r$
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 2592, room 16 
tail 0
chksum 0xef
load 0x00000000, len 0, room 8 
tail 0
chksum 0xef
load 0x00000000, len 0, room 0 
tail 0
chksum 0xef
csum 0xef
csum err
```

phoddie, 2020-08-25:
@rgujju, thank for the log. That helps. It looks like the ROM boot loader isn't successfully loading the 2nd boot loader (installed from ESP8266_RTOS_SDK⁩/⁨components⁩/⁨esp8266⁩/⁨firmware⁩/⁨boot_v1.7.bin). Here's what I see when it works:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 80MHz
SPI Mode : QIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
```

Perhaps the problem is that the SDK default flash settings don't work for your board. If you don't mind, try the following. You may need to adjust the path to boot_v1.7.bin for your system. If esptool.py isn't in your $PATH, you'll need to adjust that path too.

```
# install helloworld as usual
cd $MODDABLE/examples/helloworld
mcconfig -d -m -p esp

# reinstall the 2nd boot loader explicitly setting the flash parameters
./esptool.py write_flash --flash_freq 40m --flash_mode dio --flash_size 4MB 0 ~/esp/ESP8266_RTOS_SDK/components/esp8266/firmware/boot_v1.7.bin

# attach to xsbug
mcconfig -d -m -p esp -t xsbug
```

If that doesn't work, please capture the boot log again to see if there are any clues. Thank you,.

rgujju, 2020-09-02:
hello @phoddie 

That did the trick. Thanks a lot!

closed 2020-09-02 by rgujju