Moddable-OpenSource/moddable #385: Raise error when calling Monitor.rises() on an M5stack

jpzukin, 2020-06-15:
**Build environment:** Windows
**Target device:** M5Stack

**Description**
Raise error when calling Monitor.rises() on an M5stack.

**Steps to Reproduce**
code:
```javascript
debugger;

let message = "Hello, world";
trace(message + "\n");

button.a.onChanged = function() {
    let rises = this.rises();
    let falls = this.falls();
    trace(`button a rises=${rises}, falls=${falls}\n`);
}
```
1. Build and install the app using this build command:
`mcconfig -m -d -p esp32/m5stack`

2. Click on button a.

3. See error
[image]

**Other information**

I looked at some code.

- https://github.com/Moddable-OpenSource/moddable/blob/public/modules/pins/digital/monitor/monitor.js

```javascript
...

class Monitor @ "xs_digital_monitor_destructor" {
	constructor(dictionary) @ "xs_digital_monitor";
	close() @ "xs_digital_monitor_close"
	read() @ "xs_digital_monitor_read"
	get rises() @ "xs_digital_monitor_get_rises"
	get falls() @ "xs_digital_monitor_get_falls"

	get pwm_duty() @ "xs_digital_monitor_get_pwm_duty"
	get pwm_freq() @ "xs_digital_monitor_get_pwm_freq"
}
Monitor.Rising = 1;
Monitor.Falling = 2;
Monitor.PWM = 4;

Object.freeze(Monitor.prototype);

export default Monitor;
```

- https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/m5stack/setup-target.js

```javascript

import Digital from "pins/digital";
import Monitor from "monitor";

import AudioOut from "pins/audioout";
import Resource from "Resource";

class Button extends Monitor {
	constructor(pin) {
		super({pin, mode: Digital.InputPullUp, edge: Monitor.RisingEdge | Monitor.FallingEdge});
		this.onChanged = this.nop;
	}
	nop() {
	}
}

export default function (done) {
	global.button = {
		a: new Button(39),
		b: new Button(38),
		c: new Button(37),
	};

	global.speaker = new AudioOut({streams: 4});
	speaker.callback = function() {this.stop()};
	speaker.enqueue(0, AudioOut.Samples, new Resource("bflatmajor.maud"));
	speaker.enqueue(0, AudioOut.Callback, 0);
	speaker.start();

	//@@ microphone

	done();
}
```

This setup-target.js uses Monitor.RisingEdge and Monitor.FallingEdge, but it's undefined in the Monitor module. Probably a typo, but I'm not sure which name is better. =)

andycarle, 2020-06-17:
Thank you for the report!

You are correct that this is due to a typo in [$MODDABLE/build/devices/esp32/targets/m5stack/setup-target.js](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/m5stack/setup-target.js). It should refer to `Monitor.Rising` and `Monitor.Falling` instead of `Monitor.RisingEdge` and `Monitor.FallingEdge`.

I have committed a fix that will appear in our next open source push. Until then, you can fix the problem locally by just correcting line 9 to:

```js
	super({pin, mode: Digital.InputPullUp, edge: Monitor.Rising | Monitor.Falling});
```
Cheers,
&nbsp;- Andy

jpzukin, 2020-06-17:
Thank you for your advice. It's working as expected. =)

```javascript
debugger;

let message = "Hello, world";
trace(message + "\n");

button.a.onChanged = function() {
    let rises = this.rises;
    let falls = this.falls;
    trace(`button a rises=${rises}, falls=${falls}\n`);
}
```
[image]

closed 2020-06-17 by jpzukin