Moddable-OpenSource/moddable #386: Help with a Moddable Two app for Covid-19 Contact Tracing

mikew67, 2020-06-19:
Kind folks,
Not really a direct request for an SDK feature, but for **help on an application built with it on a Moddable Two.**
I'm helping with several volunteer groups (covid-watch.org, 600+ members, in particular) who are **building Contact Tracing apps for iOS and Android phones**.  Notably the **Covid-Watch group** has been approved by Governor of AZ as the state's approved app supplier, and is getting ready to roll out test versions of their app, complete with the necessary GAEN enablement provisions.
I'm trying to help with the development and testing effort by providing embedded system based tools for diagnostics, traffic analysis, etc.  Idea is to put together a small number to help with further development and then, if useful, a larger number to accompany more widespread testing of the apps.
So I'm proposing to put together a **GAEN Advertising Packet sniffer on a Moddable Two, with an HTTP server for receiving reports and controlling the device**.  Together with a standard USB backup battery pack it should make an inexpensive, readily producible device that we can distribute to the community.
Technically should be easy to put the software together, starting with your web server example code.  Just add Bluetooth LE scanning (and I know Bluetooth LE tech).  Designing the web-based UI might take more time, but initial implementation should be quick.
So I feel comfortable undertaking this project, but time is of the essence.  Days matter.  **So am looking for interested Moddable-qualified volunteers to help out.  Any interest?**

Mike Wirth (mike@mwirth.com)

mikew67, 2020-06-20:
So ran into my first obstacle.  Using the .../network/ble/scanner example, built for a Moddable Two (i.e., ESP32 MCU), the scanResponse object documentation doesn't appear to be complete and am having difficulty accessing its properties.  For example, the Bluetooth LE spec provides for Tx Power Level as a legal Advertising Packet data element, but it's missing from the docs (but appears to be referenced in some of the internal code).
And some of the property accessors defined in the docs don't seem to work right.
So I'm digging around in the XS source code, e.g., looking at the definition of scanResultEvent(...) in modBLEClient.c.  Heavy going for me.
So if anyone who's used the Bluetooth LE code is willing to answer some questions, let me know.

TIA,

Mike

bfriedkin, 2020-06-20:
@mikew67 -

Sorry to hear about your struggles here. Indeed the BLE Advertising Packet accessors do not cover all the available data types. I will take another pass through those to fill in more of the data types.

In the meantime, you can add an accessor for Tx Power level. In the [btutils.js](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/ble/btutils.js) module, try adding this accessor function to the `Advertisement` class:

```
	get txPowerLevel() {
		let index = this.find(GAP.ADType.TX_POWER_LEVEL);
		if (-1 != index) 
			return (new Int8Array(buffer))[index+2];
	}
```

That should hopefully return the Tx power level as a signed 8-bit value.

Please let us know if that works and feel free to ask more questions. :-)

Regards,
Brian

mikew67, 2020-06-20:
@bfriedkin 

Brian, many thanks for the quick reply.  Very helpful.  Will definitely add the getter for Tx Pwl Level (and it's a useful programming example for me to use).

BTW, not all of the AdvPkt data types defined in the Bluetooth LE spec are of the same priority.  I'll try to give you some feedback on that.

Couple functions that would be very useful are:

- One that delivers the full AdvPkt (up to its max 31 bytes) as a byte array or hex string (for hand decoding and debugging the detailed accessors), 
- One step further, breaks down the AdvPkt into an array of raw sub packets (just using the length byte at the start of each sub-pkt to parse the full AdvPkt)

- One step further (and most useful of all), parses the AdvPkt into a dictionary, with sub-pkt type (e.g., TxPwrLvl) as key and associated data as value.  (I would personally find this more useful than the individual getters.)

- (Now that I'm on a roll :-) Don't recall if there's a getter for the sender's "MAC" address that precedes the AdvPkt in any transmission, but that would be very useful and necessary for discarding duplicate packets.

Again, many thanks.  Hope to gain sufficient proficiency with the JS-to-C mechanism through this exercise to be more self-sufficient :-)

Mike

mikew67, 2020-07-01:
Brian, et al,
I've written some JS code to do some of the above, i.e.,convert the ScanResponse into an array of Adv Packet pairs: [Datatype, data] strings.  
The missing piece is the **measured RSSI** by the scanning client.  From the Newt Slack workspace/Nimble channel, I got:
`both struct ble_gap_ext_disc_desc and struct ble_gap_disc_desc have rssi field
`
Found the latter struct (but not the former) used in lines 66 and 675 of modBLEClient.c, which looks like the relevant places.  Can you or someone help me expand the ScanResult Event to include the RSSI (or similar)?

mikew67, 2020-07-02:
It looks like the critical point in the code is in the bleclient.js file, where callback events are handled. starting at line 89 and continuing with the "onDiscovered" case at line 98:
```javascript
	callback(event, params) {
		//trace(`BLE callback ${event}\n`);
		switch(event) {
			case "onReady":
				this.onReady();
				break;
			case "onSecurityParameters":
				this.onSecurityParameters(params);
				break;
			case "onDiscovered": {
				let address = new Bytes(params.address);
				let addressType = params.addressType;
				let scanResponse = new Advertisement(params.scanResponse);
				this.onDiscovered({ address, addressType, scanResponse });
				break;
			}
```
Right before the `this.onDiscovered` call, it should be possible to extract the RSSI field from the `ble_gap_disc_desc` struct embedded in the params memory block.
(Would be really nice to be able to transition from xsbug to a C debugger at this point.)

Any help here?
TIA,
Mike

bfriedkin, 2020-07-02:
@mikew67 -

Sounds like you are making good progress. Providing the measured RSSI in the scan response seems like a worthwhile addition. I will get that into our next open source update. In the meantime, something like this should work in the `scanResultEvent` function in `modBLEClient.c`:

```
...
xsmcSet(xsVar(0), xsID_addressType, xsVar(3));
xsmcSetInteger(xsVar(3), disc->rssi);
xsmcSet(xsVar(0), xsID_rssi, xsVar(3));
xsCall2(gBLE->obj, xsID_callback, xsString("onDiscovered"), xsVar(0));
```

Then in the callback:

```
case "onDiscovered": {
     let address = new Bytes(params.address);
     let addressType = params.addressType;
     let scanResponse = new Advertisement(params.scanResponse);
     let rssi = params.rssi;
     this.onDiscovered({ address, addressType, scanResponse, rssi });
```

Thank you for your other suggestions. It seems completely reasonable to access the raw scan response data from apps that want to parse the advertisement by hand, as you've already done. :-) I'll provide a simple accessor for that in the next update. Something simple like this:

```
let buffer = device.scanResponse.buffer;
let raw = new Uint8Array(buffer);
```

Does this work for you?

Thanks,
Brian

mikew67, 2020-07-02:
Brian,

Thanks.  Here's a sample output:
`Advertising Device Address: 88.48.4F.9A.08.53`
`Fields: [["FLAGS","0x06"],["MANUFACTURER_SPECIFIC_DATA",["Apple","10.05.4b.1c.97.93.60"]]]`

that I get from the following code:  
``` JavaScript
		// Parse the Advertising Packet into an array of named fields
		let fields = [];							// Parsed fields to return
		let start = 0;								// Start at first byte of data
		let stop = scanBytes.length;				// After last byte of data
		while (start < stop) {						// Loop while there's still data to parse
		let fieldLength = scanBytes[start]; 		// Length of type (1) + field data (?)
		if (start + fieldLength < stop) {			// fieldLength is in range; proceed
			let advCode = scanBytes[start+1];		// Field type code.
			let fieldTypeName = findAdvDataType(advCode);	// Convert code to type name
			let fData = scanBytes.slice(start+2, start+fieldLength+1);	// Field data bytes
			// trace("fTName: " + fieldTypeName + ". fData: " + fData + "\n");
			let fValue = fmtAdvValue(advCode,fData);	// Format data per field type
		// trace("Field: "+ fValue + "\n");
			fields.push([fieldTypeName, fValue]);	 // Push on result
			start += fieldLength + 1;
			} else {					// Specified field length is too large.
			trace("Bogus field length!\n")
			break;
			}
		}
		let report =  JSON.stringify(fields);
		// let report =  report.replace(/P/gi, '');
		trace("Fields: " + report + "\n");
```

mikew67, 2020-07-02:
Brian,

Re your suggested code additions in the scanResultEvent function, here's the code for the current version:

``` C
static void scanResultEvent(void *the, void *refcon, uint8_t *message, uint16_t messageLength)
{
	while (gBLE->discovered) {
		modCriticalSectionBegin();
		modBLEDiscovered discovered = gBLE->discovered;
		gBLE->discovered = discovered->next;
		modCriticalSectionEnd();
		struct ble_gap_disc_desc *disc = &discovered->disc;

		xsBeginHost(gBLE->the);
		xsmcVars(4);
		xsVar(0) = xsmcNewObject();
		xsmcSetArrayBuffer(xsVar(1), disc->data, disc->length_data);
		xsmcSetArrayBuffer(xsVar(2), disc->addr.val, 6);
		xsmcSetInteger(xsVar(3), disc->addr.type);
		xsmcSet(xsVar(0), xsID_scanResponse, xsVar(1));
		xsmcSet(xsVar(0), xsID_address, xsVar(2));
		xsmcSet(xsVar(0), xsID_addressType, xsVar(3));
		xsCall2(gBLE->obj, xsID_callback, xsString("onDiscovered"), xsVar(0));
		xsEndHost(gBLE->the);
		c_free(discovered);
	}
}
```

Where should the additions be made?
Thanks,
Mike

bfriedkin, 2020-07-02:
@mikew67 -

After this line:

```
xsmcSet(xsVar(0), xsID_addressType, xsVar(3));
```

Add these lines:

```
xsmcSetInteger(xsVar(3), disc->rssi);
xsmcSet(xsVar(0), xsID_rssi, xsVar(3));
xsCall2(gBLE->obj, xsID_callback, xsString("onDiscovered"), xsVar(0));
```

Regards,
Brian

mikew67, 2020-07-02:
Brian,

Thanks.  Did the trick.  RSSI is now reported to the JS world.

Mike

PS: I'll see if I can package up my code in a repo branch.

mikew67, 2020-07-02:
In BLE networking, there are several functions for construction "UUID objects".  The completeUUID16List method returns an array of these supposedly.  But how can I convert them back, e.g., to strings, hex strings in particular?

bfriedkin, 2020-07-02:
@mikew67 -

Typically there is no need to convert UUID or addresses back into strings. A scanning application can match a UUID using the `Bytes.prototype.equals` function. During discovery, if you setup a JSON service database, a characteristic/descriptor can be matched by name as well.

But sometimes you want a string when debugging. :-) I've bumped into that recently and added a new `Bytes.prototype.toString` function for tracing out string representations of the objects. This function will be available in the next open source update.

The next version of the scanner app uses this function to display the BT address of each device discovered:

```
	onDiscovered(device) {
		let scanResponse = device.scanResponse;
		let completeName = scanResponse.completeName;
		if (completeName)
			trace(`${completeName} - ${device.address}\n`);
	}
```

I've attached the `Bytes` class update for you to try in the meantime.

Regards,
Brian

[bytes.js.zip](https://github.com/Moddable-OpenSource/moddable/files/4866996/bytes.js.zip)

mikew67, 2020-07-03:
@bfriedkin -
Thanks!  Very useful.
Now while I'm at it, _brief_ documentation of the functions and classes in the btutils.js file would be _very_ useful.  Don't need full documentation for each function, but by group (e.g., the Advertisement getters) should be sufficient.

Mike

bfriedkin, 2020-07-03:
@mikew67 -

In the BLE [documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/ble/ble.md), please refer to the [Advertisement](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/ble/ble.md#classadvertisement) and [Bytes](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/ble/ble.md#classbytes) classes. Those sections will be updated in the next Open Source release to add missing functions and the new functions we've talked about here.

Regards,
Brian

dckc, 2020-10-21:
@mikew67 this sounds like a fascinating project! I wish I had noticed earlier. I'm a biomedical informatics software engineer at the University of Kansas Medical Center.

Is the embedded project still going? Is the source available? I'm looking at https://github.com/covidwatchorg but I don't see it.

mikew67, 2020-10-21:
Thanks for your interest.  The entire area of using Bluetooth LE for
Contact Tracing has morphed several times in the last couple months.  Main
issue with respect to Bluetooth LE is that the only distance measuring tool
available in the currently installed base of iPhones and Android phones is
RSSI (signal attenuation) which is an unreliable measure.  Embedded
devices, e.g,, wearables, aren't as constrained.  There's a small team of
us pursuing other options.  Let me know if you're interested and I'll link
you up with the team.

Mike

dckc, 2020-10-21:
I'm interested.

mikew67, 2020-10-21:
Send me a direct email with your email address -- we should take this off
the moddable channel.

Mike
mwirth@gmail.com

phoddie, 2020-12-09:
The specific topics here seem to have been addressed. I'm going to close this out. If there are unresolved issues here, please open theme as new requests. Thank you.

closed 2020-12-09 by phoddie

mikew67, 2020-12-10:
OK by me.  Thanks for the help.

Mike