Moddable-OpenSource/moddable #388: The sound from M5Stack is cracking/loud since ESP-IDF v3.3.2

meganetaaan, 2020-06-22:
**Build environment:** Linux
**Target device:** M5Stack Basic(esp32/m5stack)
**Moddable version** 8c76a3c10dd0e40cbfb0becf83a6b84506279edb
**Description**
With M5Stack and recent ModdableSDK build, every sound via `AudioOut` is cracking.

**Steps to Reproduce**
1. Build and install an example app using `mcconfig -m -p esp32/m5stack`.
2. Any sound including startup sound(bflatmajor) is cracking/too loud.

**Other information**

With older versions of Moddable this doesn't happen.
After some digging I found that the issue happens since upgrading ESP-IDF to v3.3.2 and [a change of i2s driver code in ESP-IDF v3.3.1](https://github.com/espressif/esp-idf/commit/70553331cc2aeeac94039c6e9f6913b94fd84b62) caused this issue.
Once I revert this commit (`I2S[i2s_num]->conf.tx_msb_right = 0;`) the sound gets normal.
They say this change is a bug-fix. But I cannot figure out if the change is a regression on the ESP-IDF side or if it need to be fixed on the Moddable side. Any workaround is appreciated.

andycarle, 2020-06-26:
@meganetaaan Thank you for the report. I'm taking a look and will report back.

&nbsp;- Andy

andycarle, 2020-07-14:
I've committed a fix for this issue that will be included in our next open source update. Thank you again for the report, @meganetaaan!

andycarle, 2020-08-06:
This fix has been pushed to open source, so I'm going to go ahead and close this issue out. @meganetaaan, please let me know if you see any further problems with this.

closed 2020-08-06 by andycarle