Moddable-OpenSource/moddable #389: MQTT over SSL blocks JS for about 2 seconds

diogoviannaaraujo, 2020-06-26:
**Target device:** ESP32

**Description**
When connecting to a MQTT server over SSL the connection is blocked for more than two seconds, no Timers are called and, if there is a lot happening, some times it does not recover from it.

**Steps to Reproduce**
Connect to a MQTT server using SecureSocket, in my case im using Google IOT Core as server. Then I setup a 10ms Timer checking Time.ticks to know how long it took  between one call and the other. After MQTT socket is open it blocks for about 2 seconds to forever.

**Expected behavior**
I expect SSL handshake to take long, because I've read that ESP32 has some hard times doing it. But I don't expect to my software to be complete irresponsive for 2 seconds at each connection.

wilberforce, 2020-06-28:
@diogoviannaaraujo 

Have you tried setting breakpoints in xsbug to see which line of code is causing the issue? I find for these sort of things if you can use a simulator to debug first before deploying to the device - I'm not sure what host you are compling with?

diogoviannaaraujo, 2020-06-28:
Just noticed that if I get the correct time using sntp the delay is not happening every socket opening, wonder why

wilberforce, 2020-06-28:
As part of the TLS negotiation, it uses the current time and I think it compares client and server.  Can you delay until you have a reply from sntp before going the MQTT set up?

diogoviannaaraujo, 2020-06-30:
If I delay the mqtt call to after the sntp return, I still get about 2 seconds block, but just once, after it disconnects and I initiate another connection everything goes fine. But if I don't fetch the time before it takes 2 seconds every reconnection.

phoddie, 2020-07-08:
Some of the crypto suites are relatively computationally complex which can take a couple seconds. It can be worse if the garage collector thrashes. You can usually see that by watching instrumentation in xsbug -- you'll see it pause updates and then show a big garbage collection count number.

One solution is to configure TLS so you use a less computationally complex crypto suite. Another is to increase the memory allocation in your manifest to reduce the number of garbage collections.

The SNTP issue observation is strange. If the time is invalid, the certificate will fail to validate and the connection attempt will fail. It will does not block or otherwise delay.:

```js
	validity = X509.decodeTBS(x509.tbs).validity;
	if (!((validity.from < now) && (now < validity.to)))
		throw new Error("date validation failed");
```

You should only use SecureSocket after you have the correct time, so that's a good change in your code

One way to try to isolate the stall is to try break in xsbug during the delay. That will require you to be a bit quick, but you can hit Command B while it is executing and it will break on the current line (ESP32 and ESP8266, at least). Then you can inspect the call stack to determine what's going on.

phoddie, 2021-07-28:
Recent optimizations to crypto functions helped speed up the handshake. It still takes some time, but it is generally improved. I'm going to close this out. If you have a specific case that is problematic, please open a new issue and reference this one. Thank you!

closed 2021-07-28 by phoddie