Moddable-OpenSource/moddable #390: Cant enable App Signing verification

diogoviannaaraujo, 2020-06-30:
Ive been trying to get App Signing Verification without secure boot to work, as described here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html#signed-app-verification-without-hardware-secure-boot

Ive added added a custom sdkconfig to my project and enabled required keys like CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT and CONFIG_SECURE_BOOT_SIGNING_KEY

Im sure the file is being loaded because I can se a Malformed warning if I write something that's is not accepted.
But I don't think it is considering those options.
It does not brake if my CONFIG_SECURE_BOOT_SIGNING_KEY is pointing to a non existing file, so I don't think he is signing anything.
It boots even if CONFIG_SECURE_SIGNED_ON_BOOT_NO_SECURE_BOOT is enabled and CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES is disabled, so I guess he is not setting the bootloader to verify what was flashed

Ive read that in 2019 signing was not supported using CMAKE with v3.3 but I was hoping that was fixed in v3.3.2 or that even if not fixed I would be able to write a app checking bootloader and then sign the image afterwards.

Am I missing something?
Thanks
Diogo Vianna

andycarle, 2020-07-09:
Hi @diogoviannaaraujo,

My understanding is the same as yours, that app signing (along with the rest of the ESP-IDF security features) does not work with CMake in ESP-IDF v3.3.2. 

I do think that the right way to proceed for your needs is to build with make instead of CMake. This is straightforward to set up with the Moddable SDK. 

You'll need to do three things:
1. Follow the ESP32 Setup steps for your target platform in an [earlier version of the Getting Started guide](https://github.com/Moddable-OpenSource/moddable/blob/c6fa046f7244847cd1ac08da0050a1f63ab140dc/documentation/Moddable%20SDK%20-%20Getting%20Started.md). If you build on macOS or Linux, you may not need to change anything from your existing setup except for defining the `UPLOAD_PORT` environment variable. If you build on Windows, you'll have several steps to follow to get MinGW set up. **You don't need to roll back to ESP-IDF v3.2.2**; this should work fine with v3.3.2.
2. Set an extra environment variable to configure the Moddable SDK to build with make instead of CMake: `ESP32_CMAKE=0`
3. Clean all of your ESP32 build output from `$MODDABLE/build/tmp/esp32` and `$MODDABLE/build/bin/esp32` and then do a fresh build. You should see make running instead of the ninja build/CMake output.

I verified that the make build is still working properly on macOS today and I have built using MinGW on Windows recently. It has been some time, but I had no problem building with Secure Boot, Flash Encryption, and App Signing using the make build on macOS in the past. 

Please give it a shot and let me know how everything turns out. If you have any further issues, just ping me here.

Cheers!
&nbsp;- Andy

diogoviannaaraujo, 2021-01-24:
I guess ill give a try now with 4.2 idf support.

phoddie, 2021-01-26:
Good luck! Please share your results.

phoddie, 2022-10-20:
Closing due to inactivity. (Also, Moddable's work shows that App Signing does work)

closed 2022-10-20 by phoddie