Moddable-OpenSource/moddable #391: MQTT over Websockets over SecureSocket Handshake Failure

zBuffer, 2020-07-06:
```javascript
const mqttConfig = {
    host: 'mqtt.naturecare.life',
    port: 443,
    path: '/',
    timeout: 30000,
    Socket: SecureSocket,
    secure: {
        protocolVersion: 0x303,
        trace: true,
        verify: false
    }
};
```

I've tried the above configuration and got the exception below:

```
> clientHello.packetize
>> handshakeProtocol.packetize
>>> recordProtocol.packetize
>>>> tlsPlainText.packetize
>>>>> tlsCompressed.packetize
>>>>>> tlsCipherText.packetize
> recordProtocol.unpacketize
>> tlsCipherText.unpacketize
>>> tlsCompressed.unpacketize
>>>> tlsPlainText.unpacketize
>>>>> alert.unpacketize
/Users/lance/Projects/moddable/modules/crypt/ssl/ssl_alert.js (82) # Break: throw!
```

alert level 2 (fatal) description 40 (handshake_failure).

Tested on device and on mac simulator -- all failed at the same spot.

phoddie, 2020-07-08:
In general TLS works fine with MQTT using `SecureSocket`. We use it with Amazon and IBM clouds regularly. I investigated an issue with CloudFlare and TLS earlier this year. As I recall, it could work but not with the default configuration. You might try stepping back to TLS version 0x302 and/or disable DHE_RSA and ECDHE_RSA (I vaguely recall when those suites are used, CloudFlare requires a crypto feature not present).

Regarding SNI, _think_ you can enable that by setting the `serverName` property in the `secure` object of the `SecureSocket` constructor.

phoddie, 2020-09-11:
Looking at the communication in WireShark shows the app sending the server name. The server rejects the connection after receiving the client hello. Here's what is sent by default:

```
Secure Sockets Layer
    TLSv1.2 Record Layer: Handshake Protocol: Client Hello
        Content Type: Handshake (22)
        Version: TLS 1.2 (0x0303)
        Length: 108
        Handshake Protocol: Client Hello
            Handshake Type: Client Hello (1)
            Length: 104
            Version: TLS 1.2 (0x0303)
            Random: 5f5aedeef6b1ab110e80ecaed70723361d4d4488724adcf0...
            Session ID Length: 0
            Cipher Suites Length: 20
            Cipher Suites (10 suites)
                Cipher Suite: TLS_RSA_WITH_AES_128_CBC_SHA (0x002f)
                Cipher Suite: TLS_RSA_WITH_AES_256_CBC_SHA (0x0035)
                Cipher Suite: TLS_RSA_WITH_AES_128_CBC_SHA256 (0x003c)
                Cipher Suite: TLS_RSA_WITH_AES_256_CBC_SHA256 (0x003d)
                Cipher Suite: TLS_DHE_RSA_WITH_AES_128_CBC_SHA256 (0x0067)
                Cipher Suite: TLS_DHE_RSA_WITH_AES_256_CBC_SHA256 (0x006b)
                Cipher Suite: TLS_DHE_RSA_WITH_AES_128_GCM_SHA256 (0x009e)
                Cipher Suite: TLS_DHE_RSA_WITH_AES_256_GCM_SHA384 (0x009f)
                Cipher Suite: TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 (0xc027)
                Cipher Suite: TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 (0xc02f)
            Compression Methods Length: 1
            Compression Methods (1 method)
            Extensions Length: 43
            Extension: server_name (len=25)
                Type: server_name (0)
                Length: 25
                Server Name Indication extension
                    Server Name list length: 23
                    Server Name Type: host_name (0)
                    Server Name length: 20
                    Server Name: mqtt.naturecare.life
            Extension: supported_groups (len=4)
                Type: supported_groups (10)
                Length: 4
                Supported Groups List Length: 2
                Supported Groups (1 group)
            Extension: ec_point_formats (len=2)
                Type: ec_point_formats (11)
                Length: 2
                EC point formats Length: 1
                Elliptic curves point formats (1)
```

Changing the TLS version and disabling DHE_RSA and ECDHE_RSA doesn't help. When a connection does succeed, do you know what the negotiated cipher suite is?

zBuffer, 2020-09-11:
I've brought down the server (sitting on mqtt.naturecare.life) in July as
our project switches to custom websockets protocol instead of MQTT over
websockets.

phoddie, 2020-09-11:
OK -- good to know. There is still some endpoint running there, but I won't try to coax it into connecting.

Should we close this issue out then, or is there something more to investigate here?

zBuffer, 2020-09-11:
I think it can be closed until we revisit this one day, or someone else
brings this up. But if you (or we) are currently motivated to check on this
issue, I can spin up stub service.

Omar-Shallah, 2021-07-02:
I have a problem connecting with a broker using ssl. 
alert code is 80 and its level 2
```

new SecureClient({
	host: "test.mosquitto.org",
	id: "moddable_" + Net.get("MAC"),
	port: 8883,
	Socket: SecureSocket,
	secure: {
		protocolVersion: 0x0303,
		certificate: new Resource("mosquitto.org.der"),
		trace: true
	},
});
```

I get the following trace

 clientHello.packetize
>> handshakeProtocol.packetize
>>> recordProtocol.packetize
>>>> tlsPlainText.packetize
>>>>> tlsCompressed.packetize
>>>>>> tlsCipherText.packetize
> recordProtocol.unpacketize
>> tlsCipherText.unpacketize
>>> tlsCompressed.unpacketize
>>>> tlsPlainText.unpacketize
>>>>> alert.unpacketize
/home/omar/Projects/moddable/modules/crypt/ssl/ssl_alert.js (82) # Break: throw!

Any Ideas ?

phoddie, 2021-07-02:
My recollection is that the mosquitto.org test server changes its certificate quite often. You may need to download the latest certificate to replace the one in the example.

Unfortunately, I can't try this myself today. If you try, please let me know what you find.

Omar-Shallah, 2021-07-04:
@phoddie I have already tried to download the certificate from mosquitto website.
Also I tried to connect to msquitto using mosquitto_pub client and it's successfully connected.

phoddie, 2021-07-04:
@Omar-Shallah - I see. I didn't realize you had tried upating the certificate.

I am able to reproduce your report. `mosquitto_sub` works:

```
mosquitto_sub -h test.mosquitto.org -t test/# -p 8883 --tls-version tlsv1.2 --cafile mosquitto.org-2.crt  -v
```

But, the example fails. The failure occurs **very** early. The client only sends the HELLO message and then the server aborts the connection. It looks like the client and the server have at least two cipher suites in common - `TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256` and `TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256`. So, presumably something else in the HELLO message is causing the failure. It used to work. So, perhaps some requirement changed at the server, such as requiring some other TLS extension. I'll investigate further.

phoddie, 2021-07-10:
@Omar-Shallah -- good news! We have a fix for the problem. It took a little while to track down, but the solution is very straightforward. We should have a push for you to try shortly. Thank you for your patience.

Omar-Shallah, 2021-07-11:
@phoddie Waiting for it. thank you.

phoddie, 2021-07-12:
@Omar-Shallah - changes are pushed. The mqttsecure has been successfully tested on ESP32, ESP8266, and the simulator.

phoddie, 2021-07-28:
Closing this issue as secure MQTT to `test.mosquitto.org` is currently working well.

closed 2021-07-28 by phoddie