Moddable-OpenSource/moddable #392: socketmbedtls memory leak

zBuffer, 2020-07-07:
**Build environment:** macOS
**Target device:** ESP32 DevKit v1

**Description**
```xs_socketmbedtls_destructor``` is missing a few mbedtls_*_free() calls, leading to memory leak.

**Steps to Reproduce**
1. Connect using SocketTLS and destroy it.
2. Repeat Step 1 for 10 times.
3. Handshake will eventually fail.

**Expected behavior**
```javascript
void xs_socketmbedtls_destructor(void *data)
{
	xsSocket xss = data;

	if (xss) {
		mbedtls_ssl_close_notify(&xss->ssl);
		mbedtls_ssl_session_reset(&xss->ssl);
		mbedtls_net_free(&xss->server_fd);
		mbedtls_ssl_free(&xss->ssl);
		
		mbedtls_ssl_config_free(&xss->conf);
		mbedtls_ctr_drbg_free(&xss->ctr_drbg);
		mbedtls_entropy_free(&xss->entropy);
		mbedtls_x509_crt_free(&xss->cacert);

		modTimerRemove(xss->timer);
		c_free(xss);
	}
}
```

Based on [mbedtls-tutorial](https://tls.mbed.org/kb/how-to/mbedtls-tutorial).

phoddie, 2020-07-08:
Thank you for the report. It is great you are able to use the mbedtls socket. It has been a long time since I've run that myself. ;) (As I recall, there were some limits to the implementation that caused me to look for alternatives)

I applied your changes, but wasn't able to build -- the code hadn't been maintained. With a few changes, it builds again.  That will be part of our next Moddable SDK update later this week.

phoddie, 2020-08-04:
Resolved

closed 2020-08-04 by phoddie