Moddable-OpenSource/moddable #393: xsCommon.h(949): error C2365: '_serialize': redefinition; previous definition was 'function'

dkurok, 2020-07-24:
Build environment: Windows
Target device: ESP32

**Description**
Following the installation-instructions and trying to build as described in Windows - Host environment setup - step6 
in https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#host-windows
I receive the following error (repeated multiple times):
nmake GOAL=debug BUILD_DIR=C:\Users\Administrator.CORP\Projects\moddable\build XS_DIR=C:\Users\Administrator.CORP\Projects\moddable\xs /c /f C:\Users\Administrator.CORP\Projects\moddable\xs\makefiles\win\xsc.mak /s
xsBigInt.c
Projects\moddable\xs\sources\xsCommon.h(949): error C2365: '_serialize': redefinition; previous definition was 'function'
Projects\moddable\xs\sources\xsCommon.h(949): error C2086: '<unnamed-enum-_Symbol_asyncIterator> _serialize': redefinition
Projects\moddable\xs\sources\xsCommon.h(949): note: see declaration of '_serialize'
NMAKE :  U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Preview\VC\Tools\MSVC\14.27.29109\bin\HostX86\x86\cl.EXE"' : return code '0x2'
Stop.
(I left out  the path before the Projects-directory for better readability)

At the end I get:
Projects\moddable\build\bin\win\release\mcconfig -m -p x-win Projects\moddable\tools\xsbug\manifest.json
'Projects\moddable\build\bin\win\release\mcconfig' is not recognized as an internal or external command,
operable program or batch file.

In 'Projects\moddable\build\bin\win\release I have serial2xsdebug.exe and simulator.exe. but not mcconfig.exe

I have Visual Studio 2019 Enterprise Edition preview 16.7.0 Preview 5 with C++ workflow installed.

**Expected behavior**
build-process should run without errors

andycarle, 2020-07-25:
Hi @dkurok,

Thank you for the detailed report. I was able to reproduce the issue using the version of Visual Studio that you specified. (This error does not occur using Community Edition 16.6.5, which is what I typically build with.)

It seems that `_serialize()` has been added as an intrinsic function in `cl` in this version of VS, causing a conflict with that enum in `xsCommon.h`. 

We're looking at our options for a long-term solution here. But a workaround is to simply remove the `_serialize` entry in that enum and the `serialize` entry in the `gxIDStrings` array in `xsCommon.c`. This should be safe for most (and perhaps all) applications, based on our initial analysis.

[This zip file](https://github.com/Moddable-OpenSource/moddable/files/4976493/xsCommon.zip) has updated versions of `%MODDABLE%\xs\sources\xsCommon.h` and `%MODDABLE%\xs\sources\xsCommon.c`. Just drop those in to replace the repo versions and then try a clean build of the Moddable SDK tools.

Please let me know if this resolves the issue.

Cheers!
&nbsp;- Andy

andycarle, 2020-07-27:
We have committed a fix for this issue that will be included in our next open source update. Thank you again for the report, @dkurok!

phoddie, 2020-08-04:
Resolved by eliminating unused `_serialize` in XS.

closed 2020-08-04 by phoddie