Moddable-OpenSource/moddable #395: Stopping WiFi access point

almini, 2020-08-05:
Hi,
my device is an ESP32 and I would like know if there is a simple way to stop a running WiFi access point?
I tried just changing the mode with `WiFi.mode = 1` but that crashes the system.

phoddie, 2020-08-05:
You have the right idea. Setting `WiFi.mode` to 1 should work. There is a bug in how the Wi-Fi state is managed. In the following code, `gWiFiState` should be set to 1 instead of 0. That eliminates the crash here.

https://github.com/Moddable-OpenSource/moddable/blob/06a3f72d15d9216e4ed3bc2f9a52e1887f3c2f42/modules/network/wifi/esp32/xs6wifi.c#L398-L400

Please let me know if that solves your problem.

phoddie, 2020-08-06:
Fix pushed. You can just update to get the change.

closed 2020-09-03 by phoddie