Moddable-OpenSource/moddable #398: Mac Catalina Host Environment Build Issue

burtonkent, 2020-08-08:
Running MacOS Catalina. I had Xcode command line tools installed, got build errors. Installed the full Xcode, but it still wouldn't build. I had to run Xcode itself and have it add some components. Am noting this here so that maybe the documentation can be updated to clarify the need for full Xcode -and- actually running Xcode to get the components needed.

Target device:  ESP8266 NodeMCU package, but right now I'm just setting up based on the [Mac Host Environment setup. I'm on step 5](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#host-mac).

**Steps to Reproduce**
1. Run make in Moddable/build/makefiles/mac

**Expected behavior**
A successful build.

**Actual Output**

This is the second run of make - it is restarting from where it last failed. I can remove the build/bin and build/tmp directories and try again if you want the full output. And I did remove those build directories based on another Github issue's suggestion.

> make
> make -f /Users/myusername/Projects/Circuits/Moddable/xs/makefiles/mac/xsc.mk
> make -f /Users/myusername/Projects/Circuits/Moddable/xs/makefiles/mac/xsid.mk
> make -f /Users/myusername/Projects/Circuits/Moddable/xs/makefiles/mac/xsl.mk
> make -f serial2xsbug.mk
> make -f simulator.mk
> make -f tools.mk
> # tools debug : xsc commodettoBitmap.js
> make[1]: /Users/myusername/Projects/Circuits/Moddable/build/bin/mac/debug/xsc: Permission denied
> make[1]: *** [/Users/myusername/Projects/Circuits/Moddable/build/tmp/mac/debug/tools/modules/commodetto/Bitmap.xsb] Error 1
> make: *** [debug] Error 2

Checking the permissions shows:
> -rw-------  1 myusername  staff  411620 Aug  8 11:20 /Users/myusername/Projects/Circuits/Moddable/build/bin/mac/debug/xsc

I did change the permission for that file to 666, it changed it back. I removed the file, it recreated it with the same permissions. Tried running using sudo, it then tried building to the wrong directory.

> /Moddable/build/tmp/mac/debug/tools/modules/commodetto/Bitmap.xsb 
shown above does not exist, but perhaps _make_ was in the process of creating it.

phoddie, 2020-08-08:
Thanks for the report. I don't have the problem running under Mojave. We do have members of the Moddable team building under Catalina without this problem. Perhaps something is configured differently. @andycarle will follow-up.

andycarle, 2020-08-10:
Hi @burtonkent,

Thank you for the report; sorry to hear that you're having trouble building.

I tried out a build on a Catalina machine tonight and I did hit a new error that is likely related to one of your notes: `xcode-select: error: tool 'ibtool' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance`. I was able to resolve that problem by xcode-selecting my full install (`sudo xcode-select -s /Applications/Xcode.app/Contents/Developer`), but that is new behavior.  We'll take a look at that and update the Getting Started document as necessary.

With that problem resolved, top-of-tree Moddable-OpenSource is building for me correctly on Catalina. I have not been able to reproduce the permissions issue that you describe: the contents of `$MODDABLE/build/bin/mac/debug` were all created with 755 permissions for me and the build completes as expected.

As Peter suggests, there must be some configuration difference at work here. Here are notes on my setup; perhaps we can spy the difference by comparing with yours:

- macOS Catalina Version 10.15.5
- Xcode version 11.5 (11E608c)
- zsh shell with no customizations beyond environment variables
- I cloned the `moddable` repository with the same user account that I am using to build. No elevated privileges/`sudo` on the clone or the build.

Also, one mild oddity I note in your build output: your repo directory is `Moddable` rather than the normal `moddable` that you should get when you clone the git repository. That should be fine (I just verified that my build completes successfully with `Moddable`), but you might double-check that your environment variables and actual path names match on case.

burtonkent, 2020-08-10:
Figured out part of it. 

- Files in /moddable/build/bin/mac/debug/ such as xsc , xsl, xsid, mcconfig, etc all had to have the execution bit set. 
- Like noted above, they were permissions 600 / -rw-------. 
- I did chmod +x /moddable/build/bin/mac/debug/* repeatedly as the build progressed. It kept failing on the permissions issue for every (non)executable file built on the previous step. 
- Changing the permission allowed make to continue the build, but would continue failing after the next executable it built and tried to use didn't have execute permissions.

I have a symlink between  /Volumes/GoogleDrive/My Drive/Projects to /Users/myusername/Projects for convenience. Am pretty sure the symlink itself wasn't the issue, the space in "My Drive" is he problem. The build was failing in two ways because of the space:

1. When starting the build while using the /Users/myusername/Projects symlink, I was able to get all the way to **make -f tools.mk** part of the build before it choked on the space.
2. If I used the actual non-symlink path with the space  (/Volumes/GoogleDrive/My Drive/Projects) it would not even complete the initial **make -f /Volumes/GoogleDrive/My\ Drive/Projects/Circuits/moddable/xs/makefiles/mac/xsc.mk** part of the build.

It looks like the path shown above was somehow determined programmatically in #, and when it tried using the non-symlink path with the space during the **make -f tools.mk** part of the build.

Supporting details for #1 and #2 above:
For #1 ran into a paths issue - 
`make[1]: *** No rule to make target `/Volumes/GoogleDrive/My', needed by `/Users/myusername/Projects/Circuits/moddable/build/bin/mac/debug/xsbug.app/Contents/Info.plist'.  Stop.`

So I went back and changed the MODDABLE environment variable from the symbolic link of /Users/myusername/Projects/Circuits/moddable to the actual directory /Volumes/GoogleDrive/My Drive/Projects/Circuits/moddable. 

Deleted the build bin/lib directories and started over. Then it fails at the beginning of the build:

`make[1]: *** No rule to make target `/Volumes/GoogleDrive/My', needed by `/Users/burtonkent/Projects/Circuits/moddable/build/bin/mac/debug/xsbug.app/Contents/Info.plist'.  Stop.`

Looks like two fixes are needed:

1. Make sure the permissions for the executables in moddable/build/bin/mac/debug/ are set properly.
2. Put the paths in double quotes so they're escaped if a space is in the directory name. 

Thanks much, I appreciate the responsiveness.

andycarle, 2020-08-13:
Hi @burtonkent,

I'm glad to hear that you've made some progress on building!

I've updated the Getting Started documentation to reflect the requirement for a full installation of Xcode and to explain how to use `xcode-select` to choose the proper installation if necessary. Thank you for bringing that issue to our attention.

I'm taking a look today at the other two issues that you raise here. I've verified the problem you describe with spaces in the `$MODDABLE` path and am exploring potential solutions. 

On the second issue, I am curious to try to understand why the Moddable SDK tools are building without the proper permissions on your system. Build output from clang generally has 755 permissions set by default... do you know if there's anything unusual about your shell or that particular file system volume that might be causing them to end up with 600 permissions?

My two thoughts on what it could be are:
     1. An unusual `umask` value on your shell. Could you try running the `umask` command before building and let me know what it outputs? One would generally expect `0022` or `000` on a macOS shell. 
     2. I gather from your paths that you are building on a mount of a Google Drive. Is there anything particularly unusual about how that is mounted?  Perhaps there is a hint to be found by running `diskutil list` or `df -h`?

I also found a bit of (old, but perhaps still relevant) info on permissions issues with Google Drive [here](https://bolesblogs.com/2013/01/23/a-warning-about-google-drive-local-file-permissions-and-web-publishing/). Are you able to run other builds on this Google Drive volume without issues? 

Thanks again!
&nbsp;- Andy

burtonkent, 2020-08-15:
Andy, that was it, Google Drive was causing issues. I've never had problems before but I'm a web developer so permissions are different for us. Thanks!

Just to be absolutely clear, your current instructions on github seem to suggest that you just need the Xcode command line tools.  If you're already developing on Mac, you probably have the Xcode command line tools installed, as I did. (Xcode command line is all that's needed for most things.) It turned out for Moddable you need to install the full Xcode app, plus to run it to add components. For the sake of people like me, please modify the Readme to say you need the full Xcode app installed. 

Thanks! About to try this out.

burtonkent, 2020-08-15:
Worked for me, thanks! Different issue now, covered in #384.

closed 2020-08-15 by burtonkent