Moddable-OpenSource/moddable #400: Semantics of PropertyDefinitionEvaluation in Object Initializer

Reinose, 2020-08-10:
### Environment
```
# uname -a
Linux reinose-ubuntu 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

# xs -v
XS 10.3.0
```

### Bug description
```
'use strict';
var x = { 42 : class   { x (  ) {  } } } ;
print(JSON.stringify(x['42']['name']));
```
The above JavaScript code produces a result as follow in xs: `""`

However, according to ECMA262 specification([http://ecma-international.org/ecma-262/11.0/](http://ecma-international.org/ecma-262/11.0/)), the code should result in: `"42"`

### Details
In [12.2.6.8 Runtime Semantics: PropertyDefinitionEvaluation](http://ecma-international.org/ecma-262/11.0/#prod-PropertyDefinition), the given code will step into **3.a** of the property definition evaluation of *PropertyName : AssignmentExpression*.
The passed argument **propKey** will be the actual name value of the result of the named evaluation of `class   { x (  ) {  } }`.
Therefore, `x['42']['name']` should be `"42"` not `""`.

NOTE: The following is the result of Node v14.5.0
```
# node -e "'use strict'; var x = { 42 : class   { x (  ) {  } } } ; console.log(JSON.stringify(x['42']['name']));"
"42"
```

phoddie, 2020-08-10:
Thank you for the report. We'll look into it.

phoddie, 2020-08-14:
We have a fix pending for this. FYI - the same behavior also occurs in JavaScriptCore.

closed 2020-09-03 by phoddie