Moddable-OpenSource/moddable #401: Spec. Mismatch: Array length property is 0 if it is defined with specific syntax

Reinose, 2020-08-10:
### Environment
```
# uname -a
Linux reinose-ubuntu 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

# xs -v
XS 10.3.0
```

### Bug description
```JavaScript
'use strict'; 
var x = [ , ... '' ] ;
console.log(JSON.stringify(x['length']));
```
The above JavaScript code produces a result as follow in xs: `0`

However, according to ECMA262 specification([http://ecma-international.org/ecma-262/11.0/](http://ecma-international.org/ecma-262/11.0/)), the code should result in: `1`

### Details
The expression `[ , ... '' ]` matches *ArrayLiteral : [ ElementList ]* in [12.2.5.2 Runtime Semantics: Evaluation](http://ecma-international.org/ecma-262/11.0/#sec-array-initializer-runtime-semantics-evaluation) and *ElementList: [ Elision_opt SpreadElement ]* in [12.2.5.1 Runtime Semantics: ArrayAccumulation](http://ecma-international.org/ecma-262/11.0/#sec-runtime-semantics-arrayaccumulation).
Since *Elision* (`,`) exists, the **length** of the array should be bigger than or equal to **1**.
However, in `xs`, the **length** of the array is always **0** if the array is initialized with *Elision SpreadElement*.

NOTE: The following is the result of Node v14.5.0
```
# node -e "'use strict'; var x = [ , ... '' ] ; console.log(JSON.stringify(x['length']));"
1
```

phoddie, 2020-08-10:
Thank you for the report. We'll look into it.

phoddie, 2020-08-19:
Fix will be included in next XS update

closed 2020-09-03 by phoddie