Moddable-OpenSource/moddable #402: Spec. Mismatch: Converting class into string does not produce proper result

Reinose, 2020-08-10:
### Environment
```
# uname -a
Linux reinose-ubuntu 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

# xs -v
XS 10.3.0
```

### Bug description
```JavaScript
'use strict';
var x = "" + class x  {  } ;
print(JSON.stringify(x));
```
The above JavaScript code produces a result as follow in xs: `"function x (){[native code]}"`

However, according to ECMA262 specification([http://ecma-international.org/ecma-262/11.0/](http://ecma-international.org/ecma-262/11.0/)), the code should result in: `"class x {  }"`

### Details
The right-hand side operand `class x {  }` is an instance of *ClassExpression*.
It is handled by [14.6.16 Runtime Semantics: Evaluation](http://ecma-international.org/ecma-262/11.0/#sec-class-definitions-runtime-semantics-evaluation).
*ClassTail* part will be covered by [14.6.13 Runtime Semantics: ClassDefinitionEvaluation](http://ecma-international.org/ecma-262/11.0/#sec-runtime-semantics-classdefinitionevaluation). (quite complicated)
The result object has **valueOf** and **toString** method and **[[SourceText]]** will be `"class x {  }"` by the step 4 in 14.6.16.

Meanwhile, the expression `'' + class x {  }` is handled by [12.8.3.1 Runtime Semantics: Evaluation](http://ecma-international.org/ecma-262/11.0/#sec-addition-operator-plus-runtime-semantics-evaluation).
Since left-hand side operand is `''` whose type is **String**, the step 7.b will be executed.
The object `class x {  }` does not have **toPrimitive** method, the step 2.g in [7.1.1 ToPrimitive](http://ecma-international.org/ecma-262/11.0/#sec-toprimitive) will be executed.

In [7.1.1.1 OrdinaryToPrimitive](http://ecma-international.org/ecma-262/11.0/#sec-ordinarytoprimitive), at first, **valueOf** method is called in the step 5.b.i.
But the result type is **Object**, so that **toString** is also called in next iteration.
Since the method is built-in function objects, [19.2.3.5 Function.prototype.toString](http://ecma-international.org/ecma-262/11.0/#sec-function.prototype.tostring) is called as a result.
Since the given object is neither a bound function exotic object nor a built-in function object, the step 2 is skipped.
In the step 3, the given object has **[[SourceText]]** and its type is **Object**.
Therefore, the step 3.a should be executed and the result should be `"class x {  }"` not `"function x (){[native code]}"`, which is a kind of *NativeFunction*.

NOTE: The following is the result of Node v14.5.0
```
# node -e "'use strict'; var x = '' + class x {  } ; console.log(JSON.stringify(x));"
"class x {  }"
```

phoddie, 2020-08-10:
Thank you for the report. We'll look into it.

phoddie, 2020-08-11:
This is not a bug. XS takes the option of not preserving the source text as allowed by the specification.

Step 3 of 19.2.3.5 Function.prototype.toString requires HostHasSourceTextAvailable(func) to be true. In XS, it is always false. Therefore. the NativeFunction representation is returned.

From [XS Differences](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20Differences.md#conformance):

>  XS does not store the source code of functions

closed 2020-08-11 by phoddie

Reinose, 2020-08-12:
Thank you for your response. I've missed [XS Differences](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20Differences.md#conformance).