Moddable-OpenSource/moddable #403: Spec. Mismatch: Optional chains are always evaluated

Reinose, 2020-08-11:
### Environment
```
# uname -a
Linux reinose-ubuntu 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

# xs -v
XS 10.3.0
```

### Bug description
```JavaScript
'use strict';
var x = undefined;
var x = x?.();
print(JSON.stringify(x));
```
The above JavaScript code produces an exception as follow in xs: `TypeError: ?: cannot coerce undefined to object`

However, according to ECMA262 specification([http://ecma-international.org/ecma-262/11.0/](http://ecma-international.org/ecma-262/11.0/)), the code should result in: `undefined`

### Details
In [12.3.9 Optional Chains](http://ecma-international.org/ecma-262/11.0/#sec-optional-chains) (especially, evaluation section), it directly returns **undefined** if *MemberExpression* is evaluated into **undefined** or **null** whatever *OptionalChain* is.
In the above code example, `x` is a _MemberExpression_ and its value is **undefined**.
Therefore, it should produce a result with `undefined`.

NOTE: The following is the result of Node v14.5.0
```
# node -e "'use strict'; var x = '' + class x {  } ; console.log(JSON.stringify(x));"
undefined
```

phoddie, 2020-08-11:
Confirmed.

It would help to provide the command line used to execute the script with `xst`. This issue, for example, occurs with the following command line:

```
xst -s ~/test.js
```

But, not with this command line:

```
xst -m ~/test.js
```

Reinose, 2020-08-12:
Thank you for the confirmation.
All the issues I've posted are executed with `xst ./sample.js`, so that it is equivalent to `xst -s ./sample.js`.

phoddie, 2020-08-19:
Fix pending

closed 2020-09-03 by phoddie