Moddable-OpenSource/moddable #407: Spec. Mismatch: Number.prototype.toString radix violation does not covered

Reinose, 2020-08-11:
### Environment
```
# uname -a
Linux reinose-ubuntu 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

# xs -v
XS 10.3.0
```

### Bug description
```JavaScript
'use strict';
print(JSON.stringify(( 42 ) . toString ( 0 ) ));
```
The above JavaScript codes produce a result as follow in xs: `"42"`

However, according to ECMA262 specification([http://ecma-international.org/ecma-262/11.0/](http://ecma-international.org/ecma-262/11.0/)), the code should throw a **RangeError**.

### Details
In [20.1.3.6 Number.prototype.toString](http://ecma-international.org/ecma-262/11.0/#sec-number.prototype.tostring), radix should be in the closed interval [2, 36].
But it is possible to use radix as 0 in xs.

NOTE: The following is the result of Node v14.5.0
```
# node -e "'use strict'; console.log(JSON.stringify(( 42 ) . toString ( 0 ) ));"
[eval]:1
'use strict'; console.log(JSON.stringify(( 42 ) . toString ( 0 ) ));
                                                  ^

RangeError: toString() radix argument must be between 2 and 36
    at Number.toString (<anonymous>)
    at [eval]:1:51
    at Script.runInThisContext (vm.js:131:18)
    at Object.runInThisContext (vm.js:295:38)
    at Object.<anonymous> ([eval]-wrapper:10:26)
    at Module._compile (internal/modules/cjs/loader.js:1201:30)
    at evalScript (internal/process/execution.js:98:25)
    at internal/main/eval_string.js:23:3
```

phoddie, 2020-08-11:
Confirmed.

phoddie, 2020-08-24:
Fixed

closed 2020-09-03 by phoddie